KMKfw/kmk_firmware #3: Support Split Keyboards

klardotsh, 2018-09-02:
- [ ] Over Bluetooth LE - the only inspiration I have here [is from Wez](https://gist.github.com/wez/b30683a4dfa329b86b9e0a2811a8c593)
- [ ] Over TRRS - suuuuuper common in QMK/TMK boards
- [ ] Over USB? - for example my Mistel Barocco does this

klardotsh, 2018-10-16:
@kdb424 ended up devising (independently of me) basically the most sane structure for matrix scans, that the refactor in #70 is moving to use for scanning the "on the same board" matrix as well: a three-byte byte array of `col_idx, row_idx, <bool>pressed`.

The "on board" matrix scan should handle flipping output/input pin indicies based on diode orientation before sending this data over the wire. For now let's just ignore "key swapping" (out of order of matrix keys) until this is handled in some more sane way in the onboard scanner (again - waiting on #70)

klardotsh, 2018-10-16:
Assigning to @kdb424 to get groundwork going eventually - however I'll probably take the Bluetooth portion of this ticket at some point.

kdb424, 2018-10-19:
Groundwork laid. Splits with uart and all configs for offsets are almost ready a3811d7 though offsets are calculated on master at this point. It will likely get converted before a PR.

klardotsh, 2018-11-09:
#82 knocked out serial, updated parent comment.

siddacious, 2018-12-05:
It would be cool look into supporting this via i2c by using [seesaw](https://github.com/adafruit/seesaw) on the slave slide. That way the slave could be pretty simple while offloading some of the work for its half.

kdb424, 2018-12-05:
I've seen that pop up several times, and conceptually seems to fit into the project, though I haven't looked into it enough to say that I have a full understanding of it.

klardotsh, 2019-07-25:
Closing this: I2C, OneWire, and BLE split support should be opened as their own tickets when/if relevant/desired/added to a milestone.

closed 2019-07-25 by klardotsh