KMKfw/kmk_firmware #39: Profile and speed up code

kdb424, 2018-10-03:
Areas that appear to need turning is the delay in pyb_hid.py, though there may be other problem areas.

klardotsh, 2018-10-12:
A huge amount of this has been done as part of the work last weekend to tune macros - on PyBoard we type at 500+ WPM on a plain ASCII text macro, and on SAMD51 we type somewhere in the 430-450 WPM range.

That said, there's probably plenty more that can be done.

kdb424, 2018-10-12:
For now, on hardware that we target, I will consider this done. This will be opened again if we end up needing support for a weaker board in the future, though with the current feature set, and no boards struggling, this is good enough. If bluetooth were to be added later, than this profile would be related to that, and this stand alone issue is no longer needed.

closed 2018-10-12 by kdb424