KMKfw/kmk_firmware #52: Target upstream CircuitPython

klardotsh, 2018-10-10:
Per conversation in https://github.com/adafruit/circuitpython/pull/1249, it'd be great to completely eliminate the KMKfw/circuitpython repository and try to track upstream commits/tags. This involves adapting and/or mainlining a few parts of KMK, particularly as it relates to the building/flashing process:

- [ ] Devise a system to replace the single frozen entrypoint `_main.py` on CircuitPython targets (this is what the upstream PR was submitting in the first place). So far the best alternative found is reading keymap off of MSC flash with a fallback on exception to a "default" or "safe mode" mapping, which feels like a solid "meet in the middle" between the projects' intentions.

- [ ] Restore MSC HID report and CircuitPython background task, which were removed in https://github.com/KMKfw/circuitpython/commit/5c606deecc93c573516c46787486cee846a6d599. This is a dependency of the above.

- [ ] Mainline MakerDiary NRF52840 MDK support? While our support of that board is very infantile and a WIP right now, mainlining this support rather than maintaining it in a fork is probably overall better for the community/ecosystem as a whole.

Thanks to @tannewt for the feedback and ideas, and feel free to use this as a scratchpad for other ways we can help to be better FOSS citizens (or at least in CircuitPython-land specifically).

tannewt, 2018-10-10:
From boot.py you can decide whether the file system is writable from circuitpython or usb: https://learn.adafruit.com/cpu-temperature-logging-with-circuit-python?view=all#selectively-setting-readonly-to-false-on-boot-3-12

We'd need to come up with a way to install a safe main.py to run during safe mode. But maybe you don't need a full safe mode, if we add the ability to select the main file to run then you could manage it on your own.

Right now you are freezing all of the code into the binary right? Is that to prevent editing or do you actually need the memory?

klardotsh, 2018-10-10:
I think I definitely like an explicit "safe mode" - could throw it behind a DIP switch or something on custom builds (some keyboards come with these anyway, if we're getting into "converting existing hardware" territory, but that's scope creep for this thread).

That's correct, all of (1) the firmware, and (2) the user keymap are frozen with `FROZEN_MPY_DIR`. At the very least this has proven necessary on Micropython targets (read: the PyBoard), and actually we're still hitting memory walls on that device (I'll let @kdb424 speak to that as he uses a handwired KMK board powered by that chip as a daily driver). I haven't done a lot of memory investigation on Atmel boards. I somehow doubt the NRF52840 will have any issues whatsoever, those things are huge.

That said, freezing the modules does allow more RAM to be free for some of the "fancier" and unique stuff we support, like emoticons baked right into the firmware and assignable to keys as macros (forgive our terrible documentation right now - example keymap [here](https://github.com/KMKfw/kmk_firmware/blob/master/user_keymaps/klardotsh/itsybitsy_m4_express/threethree.py) and implementation [here](https://github.com/KMKfw/kmk_firmware/blob/master/kmk/common/macros/unicode.py).)

I think keeping core KMK frozen is something I'd like to maintain, however user-editable code (right now this is just the keymap file, the lone entrypoint into the system) can probably move to flash. This ultimately means the user only has to reflash the entire chip (DFU/UF2/bossa/whatever, per-board) once per KMK core release (more often if they're targeting git `master` of course, but I intend to have proper versioning eventually), which seems like a far better compromise than "flash every single typo of my keymap"? Which is admittedly mostly a holdover from my background running QMK keyboards, where that is the norm (and also avoided a long-ago bug where I almost bricked a board by getting stuck in a `while True` loop and couldn't Ctrl-C it from REPL because it was an NRF board...)

kdb424, 2018-10-10:
Memory is definitely a lot more tight on the pyboard, which was our original reference board as at the time, we did not see a circuitpython board with HID and bluetooth, though that has change a lot since that board was built. As for memory limitations on the board, I have to manually garbage collect several times on that board while using macros or I will run out of memory in several places during boot within the keymap itself. This could be an issue with auto triggering of gc on micropython, though it did warn us that we were much more memory hungry than we had thought. I do not deeply understand gc on micropython or circuitpython as far as possibly tuning it, and understand that they are not the same on both platforms. I believe that on the pyboard itself we could also use SD storage in a similar way to how we can expose storage on circuitpython as well, but that would require users to add storage manually.

klardotsh, 2018-10-10:
Context worth noting: the pyboard has by a decent bit the least RAM of our supported devices at 192k

That said, with the ItsyBitsy M4 somewhat taking over as our new reference board, it's not a personal focus as much as it once was.

tannewt, 2018-10-11:
The Itsy Bitsy M4 also has 192k RAM but we do a few things with the heap that MicroPython doesn't. We split the heap into short-term objects and long-lived ones (types and modules) to improve compactness. We also collect after every import. (Video of heap behavior is [here](https://www.youtube.com/watch?v=Q_hXDPc1L8Q))

I'd love to figure out a way for KMK to work without freezing in modules because with freezing the release versioning is tied to CircuitPython's. In other words, you'll need to release a version of forked CircuitPython with every new KMK release. (Or we'll need to release CircuitPython for KMK changes.)

We also want to expose SD storage over USB but have no near-term plans to do so. (Doing so would allow us to ship Mu along with CircuitPython.)

klardotsh, 2018-10-11:
I got my "for funsies" project of the week (rotary encoder support.... [blame Reddit](https://www.reddit.com/r/MechanicalKeyboards/comments/9ma3l7/family_photo_of_a_very_atypical_variety_firmware/e7e0efg/)) out of the way with #57 , so my next thing will probably be diving into some aspect of this ticket.

So far a few ideas that are sticking out to me as priorities:

- Implementing the "try to read user's main.py, but otherwise run a fallback" logic - I'm not entirely sure yet how this would work, but I'm thinking of a text config file stored on MSC pointing to the fallback entrypoint, or perhaps just a `fallback.py` or `failsafe.py` copied over MSC. Thoughts?

- Attempting to ship KMK without _anything_ frozen: just rsync the source tree to the MSC of the device (this is now assuming a rebase to CircuitPython upstream master) and seeing what breaks

I'm still somewhat drawn to the idea of flashing KMK with frozen modules as a single entity - moving the keymap to "userspace" on MSC is a great idea (and honestly a pretty big selling point for using the firmware to begin with). Shipping KMK without frozen modules is going to require a bit more thought, mostly pending the RAM issues (I've subscribed to the issue you linked in). It may also require some changes to MicroPython itself (which we also have a fork of in this org), probably adding the GC-after-import functionality as a backport would help a ton (this is effectively what @kdb424 is doing manually in his pyboard keymap)?

klardotsh, 2018-10-11:
So far storage space itself hasn't been much of a problem, so we don't explicitly need SD storage exposed over USB yet. That said, the idea of a KMK microSD card that you can pop out of your SD reader and into any KMK-supported board with an SD reader wired up is appealing as a pipe dream goal...

klardotsh, 2018-10-12:
One more pain point I'd love to work with you on @tannewt is the recursion depth limit. Right now it looks to be at about 8 or 9ish and our bootup process teeters right on that line (in fact #61 managed to trigger it again, oops)

```python
>>> import kmk_keyboard_user
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "kmk_keyboard_user.py", line 2, in <module>
  File "kmk/entrypoints/handwire/itsybitsy_m4_express.py", line 3, in <module>
  File "kmk/circuitpython/hid.py", line 2, in <module>
  File "kmk/abstract/hid.py", line 4, in <module>
  File "kmk/event_defs.py", line 6, in <module>
  File "kmk/keycodes.py", line 9, in <module>
RuntimeError: maximum recursion depth exceeded
```

I'm pretty sure I have a patch in my current hackery branch that works around this by deferring imports to runtime rather than import/interpretation time, but it's... pretty gross.

I've been told by Dan Halbert that limit should be uppable with a configurable in the CircuitPython source somewhere, but I'm not sure if that's defined per board, per port, or globally for CircuitPython. In the spirit of trying to avoid adding things to our fork, I'd rather open the discussion about this than just hack around it in our forked builds.

tannewt, 2018-10-12:
Lots of exciting things here!

First, I'd recommend using rotaryio for CircuitPython rotary encoder support: https://circuitpython.readthedocs.io/en/3.x/shared-bindings/rotaryio/__init__.html#module-rotaryio
It uses C interrupts under the hood and was super solid when I added it.

For the main.py issue. I wonder if it just a matter of having it load different things depending on button state. It'd then essentially be a bootloader for the keyboard.

Regarding how to copy everything over, it could be like our bundle where you just copy a folder over. We've been brainstorming a gaming handheld and considered being able to load files out of a zip file. Its a nice way to bundle everything together.

What version of CircuitPython are you using? The RecursionDepthLimit error really means that you ran out of stack space. In 4.x I made it configurable and made the default less than in 3.x. Its a trade-off between stack space and heap space. You can alter the stack allocation in boot.py using [supervisor.set_next_stack_limit](https://circuitpython.readthedocs.io/en/latest/shared-bindings/supervisor/__init__.html#supervisor.set_next_stack_limit). Beware that it doesn't change the limit for the current vm (that's why I say boot.py) and that it will reduce your heap space.

Hope that helps! Very excited to see this. I'm out this weekend so it'll be a few days until I respond again.

klardotsh, 2018-10-12:
Working in somewhat scattered order here:

- The last upstream CircuitPython commit we based off of was https://github.com/KMKfw/circuitpython/commit/f7ebbacf584a1d8d0870f1221fa329e30147520b, which is now a week and some change old. So right now we haven't been working off any individual target release, but definitely commits after the 4.0.0-alpha1 tag. I'm not hellbent on always targeting that version of CircuitPython, but it's been fine for development so far (and ultimately as I move towards supporting NRF52840 keyboards, I'll need these recent commits - I discussed my work on porting the MakerDiary NRF52840 MDK based off the Feather NRF52840 source in the other ticket).

Zip file support would be really interesting (and there's somewhat of existing art in this realm by the MicroPython folks - https://github.com/micropython/micropython/pull/1797), though unless a streaming decompressor was possible, the RAM impact would likely be pretty rough on these "smaller" boards (on the NRF I guess anything goes). If there's anything I can do to help on that front (testing, or even small PRs), feel free to holler. In the meantime, copying the folder over is probably the sane way for now - it's a whole lot less complex than the other method I've come up with (which is to squash all of KMK into one huge, minified file, much like WebPack does for JS). The advantages of the latter solution would mostly be marginally faster imports, and significantly less flash usage (since each file on FAT32 has 4KB or so overhead, so even those `__init__.py` files in our tree hurt).

I'll take a look at `rotaryio` for CircuitPython targets. I'll probably end up keeping my hackish implementation around as a polyfill/alternative, particularly since M0 and anything MicroPython won't support `rotaryio`.

`main.py` I'm gonna just... think on for a bit. There's a lot going on here to digest. Might prototype a few things over the weekend.

Thanks for your interest and for jumping in to offer help! It's really appreciated! Have a great weekend!

klardotsh, 2018-10-15:
General status update on this, I've decided to take compressed modules in two steps, each probably useful on their own, but especially doing them in this order helps me wrap my head around (1) the file/compression formats in smaller chunks, and (2) the internals of CircuitPython in, again, more moderate chunks:

- Gzipped module support. This actually works in Python to some extent with my prototype at https://gist.github.com/klardotsh/cfe3038c59abbc6bb15be4cc6127497c, though it only works one layer deep as-is (unless, of course, the gzipped module handles its imports with `import_gzip` as well). I'm now working on porting this (and a very shrunken down subset of CPython's `gzip` module to expose what `builtin_import.c` will be using under the hood to end users if they for whatever reason want it - I can think of a few use cases) to C and have at least the beginnings of the implementation going places. I'll PR that upstream when it's ready (hopefully some time this week). Ultimately, `import mything` would try (in order of precedence, and ignoring frozen modules for now) directories, `.py`, followed by `.py.gz` files. This logic is already in-place in my local branch and seems to work.

- ZIP file support. My ultimate vision for this is for us to have two-file deploys of KMK: `main.py` which contains the user keymap and a simple `if __name__ == '__main__': main()` block to kick things off, and `kmk.zip`, which would be a `DEFLATE`-compressed copy of whatever snapshot of the KMK source the user wants to run. This zip file would be transparently importable as if it were just another module directory on-disk, and imports between files in the ZIP file would also work transparently.

Some core changes that will need to happen to enable KMK to use the above is that we'll need to bundle the few files we use from `micropython-lib` into the KMK ZIP (probably under import path `kmk.extlib` or `kmk.extmod` rather than expecting the files to exist at the top level of the tree, unless I instead make the ZIP file not an import path itself (this would let `kmk.zip` include top-level modules like `logging`, however this makes the ZIP importer logic _way_ more complex, because I need to then decide which ZIP or PY file has the "real" copy of, say, `logging`. This is not a can of worms we want to get into and I'm sure @tannewt likewise does not want that complexity to live in CircuitPython upstream, either).

This work does mean I'm basically out of pocket for hacking on core KMK for a bit @kdb424, but I think ultimately this is going to be a _massive_ net positive to the project so it's worth diving into upstream hackery for a bit.

klardotsh, 2018-10-19:
Closing this for the immediate time being - all officially supported functionality now targets mainline CircuitPython. When I get back to NRF stuff, I'll handle that in a different set of issues.

closed 2018-10-19 by klardotsh