KMKfw/kmk_firmware #78: RFC: Change jurisdiction of contributor CLA from BC to WA

klardotsh, 2018-10-21:
This seems like an entirely logical change now that I look that document over again and realize I used my current jurisdiction rather than my "legally a citizen of" jurisdiction in the docs.

General rationale:
- I have no idea how BC software laws work
- I have no idea how long I'll realistically reside in BC given my immigration status up here literally has "temporary" in the name
- 2/3rds of the members of this Github organization live in Washington, and the last one (myself) is still somewhat of a legal subject of Washington (I'm still a WA resident for voting purposes)
- Changing the CLA after more contributors come on board is going to be way more painful.

Need signoff/comments from all current signers of said CLA, so @kdb424 and @tannewt. If this seems sane and we make the change, we'll also all have to re-click the sign button.

Thanks all!

kdb424, 2018-10-26:
Sounds reasonable to me

tannewt, 2018-10-26:
Works for me!

klardotsh, 2018-10-26:
Updated in the Gist: https://gist.github.com/klardotsh/8f5df312a5af06998e1ba67016846fb9/revisions?short_path=955e991#diff-955e99189b8c80acce923fd3f890f778

closed 2018-10-26 by klardotsh