KMKfw/kmk_firmware #79: Re-support rotary encoders

klardotsh, 2018-10-22:
`kmk/rotary_encoder.py` has not been updated to support the new event loop/internals. Fix this, and possibly as part of this work also use `rotaryio` before falling back to our internal implementation.

pullenrc, 2021-06-11:
I have introduced rotary encoders in a different way, (without using rotaryio, mainly because it wasn't working for me) using an encoder map that will allow the use of velocity for different or combines keys. The files are in my fork, I'm not really sure how to share it other than add it here......

encoder.py
`

import digitalio
      
      
        from .kmktime import ticks_ms
      
      
        

      
      
        #********************** begin single board encoder setup *******************************
      
      
        # # use this for non-split boards, adding more action keys adds more encoders
      
      
        

      
      
        # # The encoders will populate based on several setup lists, all of the list must have the same amount of items
      
      
        # #
      
      
        # keyboard.enable_encoder = True
      
      
        # keyboard.enc_a  =    [board.D40] # list of pad a pins
      
      
        # keyboard.enc_b  =    [board.D41] # list of pad b pins
      
      
        # keyboard.encoder_count = 1 # len(keyboard.enc_a) # number of encoders based off of pad length, or use int
      
      
        

      
      
        # # encoder map follows state layering just like the key map
      
      
        # # the encoder map is a list of lists like the keymap and
      
      
        # # contains 3 item tuples in the format of (increment key, decrement key, resolution).
      
      
        # # the encoder map should have the same amount of layers as the keymap, use noop codes when you want
      
      
        # # to silence the encoder. The amount of tuples should match the number of physical encoders present
      
      
        # keyboard.encoder_map = [
      
      
        #     [
      
      
        #         (KC.VOLU,KC.VOLD,2),# Only 1 encoder is being used, so only one tuple per layer is required
      
[... 2024 lines cut ...]
        KC.LSFT,  KC.Z,    KC.X,    KC.C,    KC.V,    KC.B,        KC.N,      KC.M,    KC.COMM, KC.DOT,  KC.SLSH,  FE,
        KC.BSPC,  KC.DEL, KC.LALT, KC.LSFT, KC.LCTL,  KC.BSPC,     KC.SPC,    KC.ENT,  KC.LSFT, KC.RCTL, KC.ENT,  KC.RGUI,
        XXXXXXX,  XXXXXXX, XXXXXXX, XXXXXXX, XXXXXXX, KC.MO(1),    KC.MO(2),  KC.MUTE, XXXXXXX, XXXXXXX, XXXXXXX, XXXXXXX,
    ],
    [
        RES,      XXXXXXX, XXXXXXX, XXXXXXX, XXXXXXX, XXXXXXX,     KC.PSLS, XXXXXXX, XXXXXXX, XXXXXXX, XXXXXXX, KC.MINS,
        KC.BSLS,  XXXXXXX, XXXXXXX, KC.UP,   XXXXXXX, XXXXXXX,     KC.PAST, KC.N7,   KC.N8,   KC.N9,   XXXXXXX, KC.BSLS,
        KC.TAB,   XXXXXXX, KC.LEFT, KC.DOWN, KC.RIGHT,KC.HASH,     KC.PPLS, KC.N4,   KC.N5,   KC.N6,   XXXXXXX, KC.QUOT,
        KC.LSFT,  XXXXXXX, XXXXXXX, KC.C,    KC.TAB,  KC.UNDS,     KC.MINS, KC.N1,   KC.N2,   KC.N3,   XXXXXXX, KC.LBRC,
        KC.BSPC,  XXXXXXX, XXXXXXX, XXXXXXX, XXXXXXX, _______,     KC.SPC,  KC.EQUAL,  KC.N0, KC.RALT, KC.ENT,  KC.RGUI,
        XXXXXXX,  XXXXXXX, XXXXXXX, XXXXXXX, XXXXXXX, _______,     _______, _______,   XXXXXXX, XXXXXXX, XXXXXXX, XXXXXXX,
    ],
    [
        RES,      KC.N1,   KC.N2,   KC.N3,   KC.N4,      KC.N5,      DEBUGWL,    PRINT,   KC.N8,   KC.N9,   KC.N0,   KC.MINS,
        KC.BSLS,  KC.Q,    KC.W,    KC.LCBR, KC.RCBR,    KC.AT,      INT,     PUBLIC,    GETSET,    KC.O,    KC.P,    KC.BSLS,
        KC.TAB,   KC.A,    KC.S,    KC.LPRN, KC.RPRN,    KC.DLR,     BOOL,     DOUBLE,    KC.K,    KC.L,    KC.SCLN, KC.QUOT,
        KC.LSFT,  KC.Z,    KC.X,    KC.LBRC, KC.RBRC,    KC.PERC,    UINT ,     BYTE,    KC.COMM, KC.DOT,  KC.SLSH, KC.LBRC,
        KC.BSPC,  KC.LGUI, KC.LALT, KC.LSFT, KC.LCTL,    KC.DEL,     _______,  KC.EQL,  KC.RCTL, KC.RALT, KC.ENT,  KC.RGUI,
        XXXXXXX,  XXXXXXX, XXXXXXX, XXXXXXX, XXXXXXX,   _______,     _______, _______,  XXXXXXX, XXXXXXX, XXXXXXX, XXXXXXX,
    ]
]

if __name__ == "__main__":
    keyboard.go()
`

klardotsh, 2021-06-13:
Sharing your contributions upstream would just be a pull request in Github: https://github.com/pullenrc/kmk_firmware/pull/new/master

I haven't actually grokked all of what's happening here but I'll say in the abstract I would _love_ to work with you (and anyone else interested) to get a working rotary encoder implementation in - preferably rebased on `topic-keyboard-and-state-rebased`, but regardless. We get a lot of requests for this functionality in Matrix (@kdb424 can confirm, he handles a lot more of that room than I do) and it'd be awesome to have a "canonical" way.

pullenrc, 2021-06-14:
Awesome, I am just going to clean some things up and I send a pull request.

closed 2021-07-19 by kdb424