KMKfw/kmk_firmware #80: Support "standalone keys"

klardotsh, 2018-10-22:
This could be the push button on a rotary encoder (such as Adafruit's p/n 377), a kick pedal, whatever (though kick pedal support is a totally different type of beast than the button on the Adafruit 377). Support both of these cases as "secondary" matrices, probably scanned before (but in addition to, rather than instead of, as @kdb424 does for split keyboards) the primary matrices.

klardotsh, 2019-01-08:
I have some sketches locally of what a full-on multi-matrix KMK might look like, which is really the core of what this idea was getting at. Going to leave the issue open, but it'll probably evolve a bit to encompass a refactor of split keyboard handling as well (in preparation for some mythical v1.5/v2).

pullenrc, 2021-06-14:
I implemented a rotary encoder class(pull request just sent today), how the rotation is being handled could easily be applied to stand alone keys not in a matrix, as it's checking regular I/O lines for changes.

klardotsh, 2021-09-15:
Mostly superceded by #237 - `keypad` will provide us easy mechanisms to handle this. Not sure whether I want to leave this open or not, since I guess it's not a strictly related to moving to `keypad`, just something we'll get cheaper/free with that migration.

xs5871, 2022-04-24:
resolved by `keypad` and support for multiple scanners.

closed 2022-04-24 by xs5871