KMKfw/kmk_firmware #97: Can't macro layer switch?

kdb424, 2018-12-11:
simple_key_sequence([KC.DF(1)]) I would expect to work. Bug maybe?

klardotsh, 2019-01-08:
This was an actual (and very screwy) bug. As far as I can tell so far it seems to be fixed in #105 - @kdb424 can you test this? Shouldn't even need the `simple_key_sequence` nonsense anymore - single keys are macro targets themselves (meaning, in theory, a leader sequence should be able to trigger `KC.DF(1)` on its own, or however else you want to trigger a macro). I had a bit of a weird time (and very inconsistent luck) getting this to work on my Iris, but no issues on my Planck, so at this point I need more test data to narrow down what/where my issues are (if there's still any issues with this).

kdb424, 2019-02-23:
#105 resolved this.

closed 2019-02-23 by kdb424