KMKfw/kmk_firmware #107: Tech Debt hitlist

klardotsh, 2019-01-23:
This is a probably-ever-expanding catch-all ticket for tech debt that should be cleaned out for v1

- [ ] Remove `kmk/pins.py` entirely - this was a leftover from days of supporting MicroPython targets under a unified KMK-side API. It's gnarly, does too much, and will probably not do what the user expects if they try to use it as a generic GPIO pin in a macro.

- [ ] Clean up `ModifierKey` implementation? I dunno. It feels smelly.

- [ ] Is `kmktime` even worth keeping? Investigate what, if any, of this file is necessary in its current form vs. just importing `time` directly. Again, this file is mostly vestigial from MicroPython days

kdb424, 2019-02-21:
kmktime needs removed. Nothing of use can't be converted if it's even called any more.

kdb424, 2020-07-13:
Until the massive refactor, I believe that this was all resolved?

closed 2021-09-15 by klardotsh