KMKfw/kmk_firmware #108: Sort why not working in UIFE/Booting

kdb424, 2019-02-21:
I am unable to use my keyboard until I get to an operating system. This is very likely an upstream issue, though is definitely our problem as you would need another keyboard on hand. It's not an issue of the firmware not having time to boot as it affects booted boards upon system reboot which AFIK does not drop power from any of my keyboards therefor does not reboot the keyboard.

klardotsh, 2019-02-21:
I've confirmed a few times over that this affects my system as well - I tend to keep a QMK board on hand for days I need access to the BIOS/UEFI

klardotsh, 2019-02-21:
My first guess of where to look would be the definition order of the USB descriptors - the UEFI is probably expecting a keyboard's first USB "child" to be a HID device, and instead it's possibly getting the serial port or MSC?

kdb424, 2019-02-27:
I have tested again today, and have found that navigation once IN the UEFI does in fact work. Getting to it requires other measures, in my case, pushing a button on my motherboard, as the standard "Hold delete" did not seem to work.

elvis-epx, 2019-06-24:
It seems that putting the keyboard first on USB list is not enough, tried here by changing gen_usb_descriptor.py and rebuilding CircuitPython.

Comparing an `lsusb -v` of a JJ50 and an ItsyBitsy M4 shows that JJ50 has two descriptors: one for boot keyboard (bInterfaceSubClass=1, bInterfaceProtocol=1) and another for regular keyboard (bInterfaceSubClass=0, bInterfaceProtocol=0). The ItsyBitsy has only the second type. 

It seems easy to add the boot keyboard descriptor in gen_usb_descriptor.py but I guess it would take changes in the C modules that uses the autogenerated code as well, the rabbit hole is a bit deeper.

[itsybitsy.txt](https://github.com/KMKfw/kmk_firmware/files/3322742/itsybitsy.txt)
[jj50.txt](https://github.com/KMKfw/kmk_firmware/files/3322743/jj50.txt)
(It may be the case that boot keyboard must be present *and* first on list.)

klardotsh, 2019-07-25:
Interestingly, @kdb424 seems to have determined this is a bootup time issue - the board just doesn't boot fast enough to get into UEFI menus. If he keeps the board powered (soft reboot or backup power), and sometimes sporadically, he's able to make it in.

There's probably more debugging to be done here?

klardotsh, 2021-09-15:
These days on NRF52840 and better boards, and with all the boot time improvements that have come from CircuitPython and from our KMK refactors, I don't seem to run into this: I can usually get into UEFI and gummiboot/grub fairly easily. I'm going to close this for now as it's been inactive for a couple years and doesn't seem to be a rampant issue anymore, but we can reopen if folks are back to running into this.

closed 2021-09-15 by klardotsh

Bob837217, 2023-09-29:
I am having this issue on my KMK (latest) EK68Custom board with RP2040 and CircuitPython 8.2.6 stable.
If I get into UEFI with my K70, then navigate with EK68Custom, any key press shifts the UEFI mouse down the Y axis.  UEFI shows it detects 2 x KB and 2 x mouse.  I have the EK68Custom, K70 and mouse plugged in.  After disabling everything in boot.py except USB HID Keyboard and CDC, the UEFI shows that I have 1 x KB and 1 x mouse, and the EK68Custom does nothing on any key press.

klardotsh, 2023-09-29:
I suspect the best default for KMK going forward is likely to be disabling USB CDC and MIDI by default and locking those uses behind an "unlock mode" (I can't say "bootloader mode" because that's ambiguous - "bootloader" refers to the MCU itself). I think @kdb424 and/or @xs5871 have talked about this in the past in the context of Windows "helpfully" deciding to auto-format CircuitPython FAT drives before (which resulted in nuked BoardSource Lulus and a horrible user experience in general).

Bob837217, 2023-09-30:
Yes, I am working on a password protection during boot.py for my board right now.  Pass verified ?  enable everything Else enable only kb with boot_device=1.