KMKfw/kmk_firmware #117: Remove any mention of Proton C or Planck Rev 6

kdb424, 2019-03-08:
There is only 40k of ram on these devices, which is not enough to even load the basic firmware file. I have tested on itsy bitsy M0 express and with it's 32k of RAM just absolutely died with no chance that GC could even help.

klardotsh, 2019-03-12:
There's still a (slim) chance we could revive Frozen Module support for devices like these - where Micro/CircuitPython will read the modules straight off flash instead of loading into RAM.

That said, my energy/time availability to port Python to STM32F303 is... slim to none lately, so....

kdb424, 2019-03-12:
That does remind me that supporting things with less rom such as the Itsy Bitsy M0 may work if that were to be a thing. It's worth a shot. When that eventually happens, I'll give it a go.

closed 2019-07-25 by klardotsh