KMKfw/kmk_firmware #141: Implement BLE HID

klardotsh, 2019-08-01:
https://github.com/adafruit/circuitpython/issues/1050 is moving courtesy of @dhalbert. We should do whatever is necessary (both within KMK and, if necessary, upstream) to support connection between a single-piece keyboard or master component of a split keyboard and a BLE host (PC, phone, whatever) as soon as is reasonable. This scope does *not* include communication between two halves of a keyboard over Bluetooth.

Our HID system is fairly abstracted from the underlying implementation at this point, so a simple implemention of `kmk.hid.BLE.send` likely gets us most/all of the way there as far as targeting upstream's BLE libs. It is likely we'll want to split this HID implementation out of `kmk.hid` into an `Extension` assuming https://github.com/KMKfw/kmk_firmware/tree/topic-merge-keyboard-and-state (misnomer at this point; that branch implements the abstract `kmk.extensions.Extension` API and begins moving most non-core KMK components to use it) merges before this is worked on. We'll also need to provide workflows (how? keys? `main.py` interfaces?) to pair and bond the keyboard with the host; ideally we could support multiple host pairings (think similar to a [Logitech K810](https://www.logitech.com/en-us/product/bluetooth-illuminated-keyboard-k810) for easy swapping.

jscholes, 2019-08-12:
Any movement on this?

dhalbert, 2019-08-12:
Still in progress. There was a lot to implement in `bleio`, and now I’m debugging the Python pattt of the code.

klardotsh, 2019-08-12:
@jscholes I'm guessing you're mostly asking about adafruit/circuitpython#1050, but in case you're asking about the KMK implementation - this ticket in KMK won't be moving until (1) CircuitPython supports the underlying spec, and (2) @kdb424 and I's time/backlog frees up a bit (or someone else jumps in to write it). Speaking only for myself, I've got a pretty large refactor branch already in the works in KMK, three upstream Micro/CircuitPython PRs to see through to merge (one almost a year old - oops 😅), and an outstanding offer to @dhalbert / @tannewt  to help upstream with the underlying BLE-HID implementation which I've blocked off some time for just in case.

If I had to ballpark guess based on that and what I know my fall/winter time tends to look like, maybe November-January for a usable KMK w/BLE?

kdb424, 2020-07-13:
Did #154 resolve this?

klardotsh, 2020-07-15:
There's still a lot of UX bits I'd like to eventually tackle on this and things I'd like to explore around making BLE broadly usable and user-friendly, so I'm not sure I'd consider this ticket closed quite yet. That PR implemented a good starting point, but especially since there's no user-facing documentation about it, I don't think the feature as a whole is shipped.

uyoyo117, 2020-09-13:
any way i can help ?(if it's only doc writing) or any progress ?

dzervas, 2020-09-14:
At this point I tried and bluetooth works - but I want to lay out some goals to market it as a feature:
- [ ] Change between BT/USB (when USB connects, disable BT and reenable it later)
- [ ] Connect to multiple hosts and pivot between the active one (yes this is possible, have more than one active connections and choose where to send on each frame) - this could be done with USB as well (pivot between `USB`, `BT[0]` and `BT[1]`)
- [ ] Report battery over bluetooth with the appropriate characteristic (needs battery support for the whole firmware)
- [ ] Support LESC (is that possible in circuitpython yet?)
- [ ] Ability to erase all previous pairings
- [ ] Ability to go to pairing mode
- [ ] Change BT name from the user keymap

dhalbert, 2020-09-14:
> [ ] Support LESC (is that possible in circuitpython yet?)
CircuitPython does not support LESC yet. The Adafruit Arduino BSP for nRF just added it, though, so now we have a model implementation to copy from. But it is not high on the priority list.

dzervas, 2020-09-14:
The implementation is fairly complex and has many pitfalls - I added it just for reference

dzervas, 2020-09-15:
It seems that the fancy stuff that I wanted to do is currently not possible or too hard to implement. CircuitPython does not support interrupts yet, so I don't know when to fire an event to send the devices usage pages for example. So I can't support out-of-time device connecting/disconnecting.

tannewt, 2020-09-16:
CircuitPython's goal is to handle time critical things internally. Why do you need to send usage pages yourself?

dzervas, 2020-09-16:
No idea, that's just how I found the code - but even the adafruit's example does not do that. That's not needed, right?

dzervas, 2020-09-23:
I think with #165 #166 #167 we can close this. USB/BT coexistance isn't there yet but I think it's another feature. Battery support is another feature as well.

dzervas, 2020-09-23:
> No idea, that's just how I found the code - but even the adafruit's example does not do that. That's not needed, right?

It was not needed. Thank you @tannewt !

kdb424, 2020-10-17:
I think with #184 we can close this. It adds the initial power management. I don't want to add USB/BT coexistance in that PR as it's out of scope, and very basic power management is there. Battery reporting should also be added at some point, but I wouldn't call that critical. Full power management (deep sleep, pin wake, LESC, ect) aren't something that we can do until Circuitpython supports it, so it may be best to open issues for each of those things so they can be closed off one at a time.

kdb424, 2021-06-25:
Closing as HID support is in, and has been used for a while. We can open issues for features or bugs as needed.

closed 2021-06-25 by kdb424