KMKfw/kmk_firmware #146: Investigate why keys on layers still sometimes get stuck

klardotsh, 2019-08-20:
My example case that happens at least a few times a week on the `iris_r2`:

- VolUp is triggered with Fn2+{Key1}
- VolDn is triggered with Fn2+{Key2}

On the default layer, Key1 is itself a Fn key (`KC.MO`), and Key2 is my spacebar. VolUp will get stuck pressed if it is not released before the Fn2 key. VolDn will never exhibit such behavior.

Repro case: Press `MO(2)`, press VolDn, release `MO(2)`, release VolDn. The end result is the expected behavior, the volume down key gets released. Do this same sequence with VolUp and the volume up key never gets released, almost always resulting in dangerous volume levels on the target PC.

Not sure if this has anything to do with the keys underneath the volume functions but it's the only discerning factor I can think of so far in months of dealing with this bug.

kdb424, 2020-07-13:
If a key is "above" the layer and is held down when the layer is released, it can not be released as lifting that key will lift the key on the new "top" layer resulting in a stuck key on a layer. The protections to release all keys at a coordinate is either nonexistent or nonfunctional these days. As of the time that we started KMK, QMK had the same "bug" it their firmware.

kdb424, 2021-06-25:
Going to mark as resolved as this has not been mentioned by others. This appears to be stale. Will open a new issue if it comes back.

closed 2021-06-25 by kdb424