KMKfw/kmk_firmware #150: Tapdance no longer works.

kdb424, 2019-12-30:
Traceback (most recent call last):
  File "main.py", line 73, in <module>
  File "kmk/kmk_keyboard.py", line 300, in go
  File "kmk/kmk_keyboard.py", line 146, in _handle_matrix_report
  File "kmk/internal_state.py", line 130, in matrix_changed
  File "kmk/internal_state.py", line 139, in process_key
  File "kmk/keys.py", line 84, in _on_release
  File "kmk/handlers/stock.py", line 134, in td_released
  File "kmk/internal_state.py", line 194, in _process_tap_dance
AttributeError: 'Key' object has no attribute 'codes'

HELLA_TD = KC.TD(
    KC.A,
    KC.B,
    send_string('macros in a tap dance? I think yes'),
    KC.TG(1),
)

kdb424, 2020-07-13:
We need to see if this is even still a problem.

pullenrc, 2021-06-11:
This just happened to me, I'm not sure if I should have posted here first, at any rate I didn't see this until after the new issue was open.

klardotsh, 2021-06-13:
Reopening; closing #202 as dupe

klardotsh, 2021-06-13:
I need to get an actual KMK dev board back up to test this - I basically have my production board that hasn't been touched in years, and... somewhere in the pile of maker hardware are the MCUs and breadboards I used to test features on :)

Doing some heavy electrical and plumbing lifts on my sailboat this weekend so can't promise I'll have time to look into this before, say, mid week, but I'll try to at least get a repro so I can figure out what's going on.

If you haven't already, see if you can spin anything up on the "new master" branch, https://github.com/KMKfw/kmk_firmware/tree/topic-merge-keyboard-and-state-rebased. That's something like 95% complete and works on non-split boards (the discussion about what it'll take to get that across the finish line has been ongoing on the Matrix/Discord chat, drop in there any time if you need more real time help!)

pullenrc, 2021-06-14:
I will give the rebased branch a shot, sounds interesting.

pullenrc, 2021-06-24:
On the rebased branch TD is still giving this error, exactly as above.  I'll dig around, however, I am still familiarizing myself with the new branch.

xs5871, 2022-01-20:
I think this issue can be closed.

kdb424, 2022-01-20:
It should have auto closed if tagged. Thanks for the heads up!

closed 2022-01-20 by kdb424