KMKfw/kmk_firmware #157: Status LED Support?

Arudinne, 2020-07-08:
Hello,

I've looked over the documentation and I am not seeing anything about how to configure LEDs for Caps Lock, Num Lock or Scroll lock.

I see plenty of information on how to configure RGB leds for back-lighting and such but that isn't what I am looking for.

Is this not a supported feature or am I just missing it somewhere?

kdb424, 2020-07-08:
You unfortunately did not miss the feature, it currently does not exist. Last I remember we looked into if circuitpython supported getting that data, but that was over a year ago. Thank you for bringing this back up. We'll take a look into it.

klardotsh, 2020-07-08:
Last I heard CircuitPython has no support for this, while MicroPython does (mind you, their USB HID implementation I think varies per board and/or is only implemented on PyBoard... I don't entirely remember).

CP: Read thread starting from https://github.com/adafruit/circuitpython/issues/1009#issuecomment-409151963, and the linked PR 1074. As far as I can tell `CAPSLOCK` `SET REPORT`s are still unsupported.
MP: https://forum.micropython.org/viewtopic.php?t=6687

Arudinne, 2020-07-08:
Ah, so the issue is that CircuitPython hasn't implemented it yet.

Looks like it's still a long term goal:
https://github.com/adafruit/circuitpython/issues/2348

thinghacker, 2021-03-27:
Looks like this has been addressed in https://github.com/adafruit/circuitpython/pull/3302 and is now in the main branch of circuitpython so perhaps kmk could support status LEDs now?

TobiasWegener, 2021-12-13:
Hi @Arduinne ,
so was wanted to have a Numlock (`KC.NLCK`) status indication, I came up with the following little solution (hacky):
```
import board
import digitalio
from kmk.kmk_keyboard import KMKKeyboard
from kmk.keys import KC
from kmk.handlers.sequences import send_string

keyboard = KMKKeyboard()

def NumStatusLED(*args, **kwargs):
    '''When called reverses the bool state of LED giving status'''
    LR1.value = not LR1.value 

KC.NLCK.after_press_handler(NumStatusLED)  # This triggeres the LED after the Keypress

# Define pins for the LED over digitalio
C1 = digitalio.DigitalInOut(board.GP18) 
C1.direction = digitalio.Direction.OUTPUT
LR1 = digitalio.DigitalInOut(board.GP2)
LR1.direction = digitalio.Direction.OUTPUT

if __name__ == '__main__':
    C1.value = 0  # sets the GPIO pin to ground
    LR1.value = 1 # sets the pin to 3.3V for start condition - LED ON
    keyboard.go()
```
If there is a smarter or better way to  do it, I am very interested!
I have a matrix of LEDs with Columns and Rows, that's why needed the GPIO to pull to ground instead of using the Ground pins directly.

This at least work out of the box for me with Cuircitpython 7 and an up-to-date version of kmk.
I hope that is helpful.

Arudinne, 2021-12-14:
@TobiasWegener,

I've tabled the project I was working on at the time. Glad to see it should work in newer versions of CircuitPython whenever I have time to work on it again. Thanks!

Jambul46, 2021-12-27:
hi is this still unavailable in the KMK?

klardotsh, 2021-12-30:
> hi is this still unavailable in the KMK?

yes. to my knowledge nobody has submitted a PR to implement this, and kdb424 and I have both largely stopped contributing new features ourselves, so I believe status LEDs remain unimplemented.

Jambul46, 2021-12-30:
oh ok but still thanks for reply

TobiasWegener, 2021-12-31:
@Jambul46 Still you can use the workaround with the `Key.after_press_handler(ChangeStatusLED) ` [example code](https://github.com/KMKfw/kmk_firmware/issues/157#issuecomment-992716323), which does work fine in KMK.
@klardotsh do you think it would be useful to add this workaround to the documentation?

Jambul46, 2021-12-31:
oh thank's that didn't worked for me first time i tried (i forgot to add keymap)
and to make it work with capslock i only have to change KC.NUMLOCK (KC.NLCK) to KC.CAPSLOCK right?

if am right it will turn the led on even when the computer start's with numlock disabled right? (if yes it's no big deal for me but i may be kinda annoying)

TobiasWegener, 2022-01-01:
@Jambul46 Yes, that should do the trick.
It should look something like this for your case:
```
import board
import digitalio
from kmk.kmk_keyboard import KMKKeyboard
from kmk.keys import KC
from kmk.handlers.sequences import send_string

keyboard = KMKKeyboard()

def caplck_status_led(*args, **kwargs):
    """When called reverses the bool state of the LED pin."""
    LR1.value = not LR1.value 

KC.CAPSLOCK.after_press_handler(caplck_status_led)  # This triggeres the LED after the Keypress

# Define pins for the LED with digitalio, with negativ from LED on ground of the controller
LR1 = digitalio.DigitalInOut(board.GP2) # change to pin of LED
LR1.direction = digitalio.Direction.OUTPUT

if __name__ == '__main__':
    LR1.value = 0 # sets the pin to 0 V for start condition - LED OFF = Capslock off
    keyboard.go()
```
The LED starts in the "off" condition.

This should work as expected.

Jambul46, 2022-01-01:
Yup already did that

TobiasWegener, 2022-01-03:
@Jambul46 and does it work as expected?

Jambul46, 2022-01-03:
Yes, but like i said if computer turns on with numlock disabled it still lights up

xs5871, 2022-01-19:
I created a PR for an extension that will enable you to read lock states as set by the host #302.

Jambul46, 2022-01-19:
Thank's man gonna try that tomorow

Jambul46, 2022-01-21:
what board do you use? it doesn't look like it's working on my pi pico

xs5871, 2022-01-21:
also rp2040. After a couple of days of use, I noticed that it doesn't update the lock status when the device gets handed over from BIOS to OS, but once you change the locks again when the OS is up, everything is working fine again. Can anyone else confirm this?

Jambul46, 2022-01-21:
well it doesn't even compile for me

Jambul46, 2022-01-21:
or better said it doesn't run even when i try the kmk folder from your fork

Jambul46, 2022-01-21:
and the original kmk folder also doesnt work something it keys.py __getvar or somethig like that

xs5871, 2022-01-21:
what version of circuitpython are you on?

Jambul46, 2022-01-21:
the latest i think and the error is

Traceback (most recent call last):
  File "code.py", line 137, in <module>
  File "kmk/keys.py", line 364, in __getattr__
ValueError: Invalid key

and when i comment out tap dance part it work's

Jambul46, 2022-01-21:
ps: using the original code with the "dumb" numLock and capsLock detect

Jambul46, 2022-01-21:
and when i use you'r example with the official KMK i get this

code.py output:
Traceback (most recent call last):
  File "code.py", line 3, in <module>
ImportError: no module named 'kmk.extensions.lock_status'

Jambul46, 2022-01-21:
so tapDance work's now i was missing few lines but the status lock still doesn't

xs5871, 2022-01-21:
The PR hasn't been merged into the master branch yet; that would explain the ImportError.
Can't say anything about the other errors you're seeing.

Jambul46, 2022-01-21:
the error i get now is Traceback (most recent call last):
  File "code.py", line 165, in <module>
NameError: name 'led_cap_lock' is not defined

Jambul46, 2022-01-21:
https://pastebin.com/23qkW6Kb my current code (the lock detect part doesn't work but is saves without error)

xs5871, 2022-01-22:
Please consult the matrix channel for this level of support.

Jambul46, 2022-01-22:
matrix part is fixed now only the led status

closed 2022-04-16 by kdb424