KMKfw/kmk_firmware #191: update status of hardware support in README.md?

mkarikom, 2020-11-09:
Thanks for publishing this awesome project!
I'm confused by lines 60-64 in [README.md](https://github.com/KMKfw/kmk_firmware/blob/master/README.md):

```
  > \* The ItsyBitsy M4 Express is the only controller we currently support in
  > non-handwired configurations, using our [ItsyBitsy to Pro Micro converter
  > PCB](https://github.com/KMKfw/kmk_firmware/tree/master/hardware) designed by
  > @siddacious and @kdb424. It is our most-recommended MCU until [the ItsyBitsy is
  > updated with an nRF52840
```

The ItsyBitsy nRF52840 is now available [here](https://www.adafruit.com/product/4481).  

Will the ItsyBitsy Pro Micro converter also work with this board?

Thanks!

kdb424, 2020-11-09:
It will in fact work with that as far as I can tell. It's untested, but I don't see any changes that would cause any issues. The readme is getting an update in #189 and our most popular microcontroller will be the nice!nano https://nicekeyboards.com/ as it does not need the adapter, as well as havig a charge circuit onboard.

mkarikom, 2020-11-10:
awesome thanks!

closed 2020-11-10 by kdb424