KMKfw/kmk_firmware #194: Mouse HID partially implimented

PanHandle, 2021-03-23:
Mouse emulation seems to be supported in `kmk/hid.py` but is not implemented elsewhere.
```
hid.py:16:    MOUSE = 2
hid.py:23:    MOUSE = 0x02
hid.py:30:    KEYBOARD = MOUSE = SYSCONTROL = 0x01
hid.py:35:    HIDReportTypes.MOUSE: 4,
hid.py:209:            if up == HIDUsagePage.MOUSE and us == HIDUsage.MOUSE:
hid.py:210:                self.devices[HIDReportTypes.MOUSE] = device
```
Keyboards with track-point input or mouse-keys cannot be created without additional controllers.

Is the feature undocumented or not possible?

kdb424, 2021-03-25:
That was left as a stub to make it easier to add the feature in, however, it is not currently in place.

RichardBronosky, 2021-04-10:
`git log -u` suggests a storied history of attempted mouse support. I'd like to see this get finished someday. Maybe I'll learn enough to help.

joshreve, 2021-09-11:
Curious as to the progress here.  I have a [repo for a Dactyl-Manuform variant](https://github.com/joshreve/dactyl-keyboard) which can support OLED, trackball, and hopefully PCBs with per key RGB longer term.  All the current QMK compatible controllers don't have enough I/O to support this, but a Pi Pico or possibly an adafruit feather does.  I am looking at working on this to achieve the full array of functionality in one controller.  My main target is the Pico due to the outstanding performance to cost ratio.  

I am looking to add both mouse keys and PMW3360 support.  Theoretically this can extend to other modules like the ADNS sensors with additional work.

If it would be possible to connect to get more information on the gaps to achieving this, I would like to start adding capabilities.  I am a seasoned engineer an system architect and this is currently my primary hobby project.  I have tons of experience with Python and pythonic methodologies so all I need to get started is a basic layout of what is missing and where the project prefers these capabilities to be implemented.

kdb424, 2021-09-12:
@joshreve Join is in the Matrix, or Discord (they are linked in the main readme) and we can chat about whatever you need in order to get you started. Would love to help you get that made.

closed 2022-03-19 by kdb424

henrebotha, 2022-03-19:
Why is this closed?

kdb424, 2022-03-19:
@henrebotha Mouse keys have been added, and trackpoint support is already being worked on as well. https://github.com/KMKfw/kmk_firmware/blob/master/docs/mouse_keys.md

RichardBronosky, 2022-03-21:
> @

Woot! This is great news. Thank you for the update. And a huge thanks to everyone involved in finally getting this into the trunk! ☮️❤️🌈