KMKfw/kmk_firmware #197: Matrix and Discord invites don't work

AndreEagleDolla, 2021-04-20:
I am trying to join the matrix/discord server but the discord invite is expired and the matrix rooms all say:

"#kmk-support:kmkfw.io is not accessible at this time. Try again later, or ask a room admin to check if you have access."

closed 2021-04-21 by AndreEagleDolla

klardotsh, 2021-04-23:
You're not the first person to mention getting sent to the old `kmkfw.io` homeserver but I can't for the life of me figure out where this link is - do you happen to remember what sent you there? From what I can tell the current `master` README has working links for both Matrix and Discord

AndreEagleDolla, 2021-04-25:
Absolutely! I found it under docs > support https://github.com/KMKfw/kmk_firmware/blob/master/docs/support.md

klardotsh, 2021-05-03:
Thanks! Fixed in `master`

commit 01daa7a0 (2021-05-03): fix: resolve #197, remove broken Matrix links and instead duplicate README blurb