KMKfw/kmk_firmware #198: Change UART pins definition

Max585t, 2021-04-30:
This would allow for better split keyboard support. Currently, when using Raspberry Pi Picos in split keyboards, there is no way to have either side be the target because of the way Rx and Tx pins are set. 

The code would change from
```python
def init_uart(self, pin, timeout=20):
        if self.is_target:
            return busio.UART(tx=None, rx=pin, timeout=timeout)
        else:
            return busio.UART(tx=pin, rx=None, timeout=timeout)
```
to
```python
def init_uart(self, pin, timeout=20):
        if self.is_target:
            return busio.UART(tx=None, rx=pin[1], timeout=timeout)
        else:
            return busio.UART(tx=pin[0], rx=None, timeout=timeout)
```

The users keyboard definition, you would have to define the uart pins as such:
```python
keyboard.uart_pin = (board.TX, board.RX)
```

kdb424, 2021-06-25:
If this was tested on actual hardware, feel free to PR this. We can validate against other hardware to ensure it won't break other boards.

Max585t, 2021-06-25:
I have tested this on a pi pico. I would like to make a PR but I cannot publish my branch

kdb424, 2021-06-25:
I'll see about getting a PR then based on this. I have 2 Pro Micro 2040's coming to test this now as well. Thank you for solving this.

Tonasz, 2022-02-07:
Isn't this issue solved as a byproduct of https://github.com/KMKfw/kmk_firmware/pull/301? It changed target assignment and added `uart_flip` parameter (disables unnecessary pins swap when False)

closed 2022-03-19 by kdb424