KMKfw/kmk_firmware #204: Unable to flash CircuitPython to MakerDiary nRF52840 MDK (NO USB dongle)

teoremma, 2021-06-14:
I am trying to run this project on the suggested board [MakerDiary nRF52840 MDK](https://makerdiary.com/collections/frontpage/products/nrf52840-mdk-iot-development-kit). I am encountering issues flashing the official CircuitPython `.hex` provided [here](https://circuitpython.org/board/makerdiary_nrf52840_mdk/). I have tried drag-and-drop to the `DAPLINK` mass storage that the MDK provides as well as `pyocd -t nrf52 circuitpython.hex`. I both cases it appears to have flashed succesfully but no `CIRCUITPY` mass storage appears afterwards.

As per #52 I understand that @klardotsh added the initial support for this board in CircuitPython so I hope I can get some useful pointers here.

Anybody has any ideas on why this could be happening? Any workarounds?

klardotsh, 2021-06-20:
While I think I still own that hardware somewhere in a box, I haven't touched the MDK in years, and don't really maintain the upstream board definitions anymore. Have you asked for general troubleshooting assistance [in the Adafruit Discord](http://adafru.it/discord)?

klardotsh, 2021-06-20:
Closing because this isn't a KMK bug, though I'll still respond as time allows.

@teoremma here's the docs I wrote about flashing that device back in 2018: https://github.com/adafruit/circuitpython/tree/main/ports/nrf/boards/makerdiary_nrf52840_mdk. It might still apply today.

closed 2021-06-20 by klardotsh

teoremma, 2021-06-26:
Thank you. I was able to successfully flash it following the steps [here](https://community.makerdiary.com/t/nrf52840-mdk-iot-development-kit-and-circuitpython/312).