KMKfw/kmk_firmware #218: Quick install guide for Raspberry Pico (may work with other boards)

elric91, 2021-07-22:
1. Install CircuitPython on the board (https://circuitpython.org/board/raspberry_pi_pico/)
2. Get KMK from the master branch (https://github.com/KMKfw/kmk_firmware/archive/refs/heads/master.zip)
3. Copy the KMK folder and the boot.py file at the root of the CIRCUITPY usb device
4. Create a new file on the root of the CIRCUITPY usb device with the following content : 

```
print("Starting")

import board

from kmk.kmk_keyboard import KMKKeyboard
from kmk.keys import KC
from kmk.matrix import DiodeOrientation

keyboard = KMKKeyboard()

keyboard.col_pins = (board.GP0,)
keyboard.row_pins = (board.GP1,)
keyboard.diode_orientation = DiodeOrientation.COLUMNS

keyboard.keymap = [
    [KC.A,]
]

if __name__ == '__main__':
    keyboard.go()
```

5. With a wire / paperclip / whatever, connect GPIO 0 & GPIO 1 (https://learn.adafruit.com/getting-started-with-raspberry-pi-pico-circuitpython/pinouts)

6. If it prints a "A" (or a "Q" or ... depending on your keyboard layout), you're done !

elric91, 2021-07-22:
tutorial. Not a problem

closed 2021-07-22 by elric91

klardotsh, 2021-07-22:
Is there anything here you think might be helpful to add to https://github.com/KMKfw/kmk_firmware/blob/4eba21a2fdf024d99a29f7f8f00b0f4f1cd36ad9/docs/Getting_Started.md?