KMKfw/kmk_firmware #236: FEATURE REQUEST - Possibility for compile time option to remove ALL USB Enumerations other than keyboard?

bleughb, 2021-09-14:
I have an embedded linux system that won't enumerate a KMK keyboard. with a PI PICO
it works perfectly with QMK and 32u4
and i've just discovered PRK firmware can also work with it if it's compiled witouth the mass storage option
It won't enumerate due to the extra things it reports itself as over USB. If it were just a plain old keyboard over USB

kdb424, 2021-09-14:
It's not possible to do at compile time, though Circuitpython 7 seems to be able to do this. It's worth tinkering with, and reporting back.

klardotsh, 2021-09-15:
While CDC and MIDI (and, uselessly to any wired KMK board, HID) can all be disabled, MSC cannot be disabled through any officially (by CircuitPython) exposed APIs.

- https://circuitpython.readthedocs.io/en/latest/shared-bindings/usb_cdc/index.html#usb_cdc.disable
- https://circuitpython.readthedocs.io/en/latest/shared-bindings/usb_midi/index.html#usb_midi.disable
- https://circuitpython.readthedocs.io/en/latest/shared-bindings/usb_hid/index.html#usb_hid.disable

In ancient history, KMK actually did ship a CircuitPython build with MSC disabled (in effect meaning the flashing sequence was very QMK-ish). That could still be done via a custom build of KMKPython, but I think the most I'd like to invest there is maybe some docs on how to do such a modification yourself, with a giant `NO WARRANTY, AT YOUR OWN RISK` label slapped on there. This seems unlikely to impact many KMK users and the process for building Circuit/KMK Python without MSC is going to be pretty unpleasant for most folks.

bleughb, 2021-09-15:
Hi, Agree, building would be unpleasant for most users - I've had a month long attempt at finally getting PRK to compile, including a solid 2 week / many hours per day - primarily as i'm a bit of a noob to linux, and all this really. as long as there's a bit of a documented process to get that UF2 file out of the thing - that''d make a few of us real outisder use cases happy!.
Personally, i'm stubborn enough to muddle through, and eventually to help others a bit by blogging a bit the muddling.
This application is actually for a emulation device using an allwinner chipset inside - it (and others) don't like anything that has aything other than vanilla USB HID...
**THANKYOU** for at least considering this 'out there' idea...i'm sure it'll be useful for more than just myself. I only discovered the whole QMK (and now PRK) thing because the standard Arduino libraries wouldn't enumerate correctly....and i've read various forums over the year to see that a few (but not many) have had similar issues over the years!
if I can help in any way, please yell, I'll be slow, time poor and 'dumb' but i'll do what i can

mhorowitz, 2021-12-07:
In Circuitpython 7.1.0, this is pretty easy.  In boot.py:
```
usb_cdc.disable()
usb_midi.disable()
storage.disable_usb_drive()
usb_hid.enable((usb_hid.Device.KEYBOARD,), boot_device=1)
```
This is working ok for me, with one caveat: it doesn't work at boot time on my macs (see #292).  Note that if you do this unconditionally, you can no longer easily modify your code.  The simplest way around this is to enter [safe mode](https://learn.adafruit.com/welcome-to-circuitpython/troubleshooting), but you can also programmatically disable this logic (say if a key is pressed) if access to the board is inconvenient.

idesignstuff, 2022-02-22:
As a user who wants to attach a home built, hand wired macropad to a corporate-owned windows computer with lots of nannies, I'd love to be able to default to only enumerate hid, and alternatively reboot it with a key pressed to allow mounting as a drive again for that boot.

kdb424, 2022-02-22:
I believe that this is still not fully possible, and upstream related. If I am incorrect, please submit a pull request to KMK itself, even as a draft, on how this may be done, and we,lol work with you to get it integrated.

mhorowitz, 2022-02-23:
@idesignstuff take a look at #292, particularly the [boot.txt](https://github.com/KMKfw/kmk_firmware/files/7665718/boot.txt) attached there.  This uses digital i/o at startup to switch between serial+storage mode, and keyboard-only mode.  (The boot device feature doesn't seem to work, and I think that might be an upstream issue, and should not affect your use case.)  You'll probably need to customize the choice of i/o pins for your custom board, and also disable cdc, but the general technique should work for you.

kdb424, 2022-03-19:
This is in the process of being documented, or already documented.

closed 2022-03-19 by kdb424

idesignstuff, 2022-03-19:
Yep, I tried it and it works great. If it's just plugged in, no storage. If I press a key while it boots, storage! You guys are magical.