KMKfw/kmk_firmware #244: One Shot Keys (Feature Request)

ldsands, 2021-09-17:
I don't see anything like the One-Shot Keys feature from QMK. Since I started using it, my wrist pain has really been helped by this feature. QMK's documentation for this can be found [here](https://docs.qmk.fm/#/one_shot_keys?id=one-shot-keys).

A brief description of this (taken from QMK's website): "One shot keys are keys that remain active until the next key is pressed, and then are released. This allows you to type keyboard combinations without pressing more than one key at a time. These keys are usually called “Sticky keys” or “Dead keys”."

This would include the modifier keys including `ctrl`, `win` or `super`, `alt` and `shift`. It could also include one shot layer switching as well. I'm sure I'm forgetting something.

My python experience is limited to (mostly) self-taught data science so I doubt that I would be able to help with this, but I would love to see this implemented into KMK. Thank you in advance to anyone who helps with this (and KMK in general)!

kdb424, 2021-09-19:
We are currently blocked by #222 for combos, but once that is solved, we can work on combos.

xs5871, 2022-02-17:
This issue can be closed.