KMKfw/kmk_firmware #276: It's hard to use my own matrix scanner

mhorowitz, 2021-11-15:
I'm playing around with KMK on a custom board which is using a MCP23018 i/o expander.  There is existing code in https://github.com/KMKfw/kmk_firmware/blob/b287b2112c1a730fb2ad470f02c05ae96160a1de/kmk/matrix.py to handle this chip, but the mcp.get_pin api is inefficient, as I end up reading from the i2c bus for each input.  I wrote my own scanner which does one i2c read operation.  However, it's difficult to use with kmk_keyboard, because _init_matrix always uses the fixed class.

It looks like the intent is to use the matrix_scanner attribute, and https://github.com/KMKfw/kmk_firmware/pull/142 did that, but https://github.com/KMKfw/kmk_firmware/commit/9821f7bcc352dc7de1a59bb3d28c98714fbe15b1 (probably accidentally) undid it.

If it would help to submit a PR to redo #142, I can submit one.

klardotsh, 2021-11-15:
This is related to (and probably going to be superceded by) https://github.com/KMKfw/kmk_firmware/issues/237

mhorowitz, 2021-12-06:
Yeah, I can see that. But #262 looks to be a WIP, and this is a one-line change.

klardotsh, 2021-12-06:
I hadn't read #142 again to see that it was such a simple one-liner. implemented as a yolo commit to master, thanks for reporting!

commit 8063898e (2021-12-06): fix: resolve #276 (reimplement #142) by allowing custom matrix scanners
closed 2021-12-06 by klardotsh