KMKfw/kmk_firmware #278: Permissive hold

sadekbaroudi, 2021-12-01:
First and foremost, great project! I'm very pleased with how quickly and easily you can write firmware!

I am a fairly fast typist, and I use home row mods. Without a feature like permissive hold it is very hard for me to use any keyboard using kmk.

If I set the tap_timeout too low (100 ms or below), I trigger the hold while typing. If I set it longer, I have to slow down my typing when using any mods or layers to a point of being very inconvenient.

When using QMK or ZMK, they have a mechanism that allows me to resolve this.

This works perfectly in QMK:
https://beta.docs.qmk.fm/using-qmk/software-features/tap_hold#permissive-hold

ZMK has a clear outline of how they select which:
https://zmk.dev/docs/behaviors/hold-tap/

Both of these allow you to configure it per key as well. For instance, I need the shift to be very fast, and the others can be a bit slower.

klardotsh, 2021-12-01:
I personally don't use this functionality in other firmwares so I likely won't be taking this up myself, but I'll leave the ticket open in case someone else is perusing through issues looking for inspiration some time and wants to submit a pull request.

xs5871, 2021-12-02:
_Just_ hold-preferred / tap-preferred, asignable per key, is a fairly straight-forward implementation.
We could extend the validator signature by a `behavior` stored in meta and do something like this for configuration:
```python
from modules.holdtap import InterruptBehavior
FASTSHIFT = KC.MT(KC.X, KC.LSFT, behavior=InterruptBehavior.HOLD)
```
Then, just launch the corresponding handler in `ht_activate_on_interrupt` based on the `InterruptBehavior` enum.
 
It would be nice to also have the other hold-tap flavors as layed out by the ZMK reference. Those are somewhat more involved.

xs5871, 2022-02-17:
The HoldTap enhancements solve this issue.

closed 2022-04-16 by kdb424