KMKfw/kmk_firmware #279: using mod tap, holding two mods at the same time doesn't work

sadekbaroudi, 2021-12-01:
I use home row mods (so, mod tap), and I cannot hold two of them at the same time.
https://github.com/sadekbaroudi/kmk_firmware/blob/master/boards/ffkb/main.py#L93

For instance, if I want to press control alt delete, it just doesn't work.

What I do is press and hold my A and S keys at the same time (mod tap ctrl alt), and it outputs whichever key I pressed first, and then does nothing.

xs5871, 2021-12-02:
Appears to be fixed by PR #281.

pcharest2000, 2021-12-08:
The feture works only if you tap the mod in sequence...   slowly once after the other once the timeout is homored, .   very annoying hope we get a fix for this

pcharest2000, 2021-12-10:
> Appears to be fixed by PR #281.

I've pulled your repo and does not seem to fix combined modtaps :( ... will look    into the master branch and see whats going on

closed 2022-01-18 by klardotsh

sadekbaroudi, 2022-01-18:
The previous comment indicates that it's not fixed by #281 - If this is true, shouldn't this stay open?
@pcharest2000 - can you confirm?

pcharest2000, 2022-01-19:
> The previous comment indicates that it's not fixed by #281 - If this is true, shouldn't this stay open? @pcharest2000 - can you confirm?

has been working for over 3 weeks,  I will pull from the master branch and retest to confirm by tommorow...

pcharest2000, 2022-01-22:
> > The previous comment indicates that it's not fixed by #281 - If this is true, shouldn't this stay open? @pcharest2000 - can you confirm?
> 
> has been working for over 3 weeks, I will pull from the master branch and retest to confirm by tommorow...

I have tested  this morning  from the master branch sorry for the delay  it works on my side as per latest +- 4 weeks I have been using....    And confirm it works with 2 and 3 keys modtap combination but read further.   Please note that the behavior does not seem to be identical to QMK... If I do press combination of modtap rapidly  (ALT-CTRL) sometimes the  first key hit to the  bounded   modtap does get sent...   example if a is modtap to SHIFT and s to CTRL...  Pressing a and s  almost simultaneously  sometimes  a  is sent instead of SHIFT-CTRL...  But if you tap a than s  in a sequence the proper modtap gets sent (within the timeout)....   At least for now MODTAP is partially functional contrary to before the pull but still needs some work, but it is much better.   Thus it still needs some work.

xs5871, 2022-01-22:
If I understand correctly, the issue you're describing is related (or identical) to #278.

pcharest2000, 2022-01-22:
> If I understand correctly, the issue you're describing is related (or identical) to #278.

After more testing  tap_timeout  value between 100  and 300  does not seem to change the  behavior you really have to tap_mod the keys in a sequence and not simultaneously.    With high value like 1000  it becomes more evident that something is not working:   u have to wait for the timer to trigger and than hit the second key....

PR #281 (2022-01-18): make TapDance a module
Probably a necessary or at least helpful step to unifying multi-purpose keys (TapDance, ModTap, etc...), see #222.
It doesn't yet implement the suggested API, but it adds a new module method `Module.process_key` that allows for intercepting and potentially modifying any `Key` before it is handled by `KMKKeyboard.process_key`.
It also became apparent that the `TapDance` code/logic would need a complete rewrite, ideally something similar to `HoldTap` with a proper state machine. Current code is difficult to follow and debug.

* fixes #279