KMKfw/kmk_firmware #285: KC.NO resolves to KC.N

kdb424, 2021-12-05:
`KC.NO` seems to be read as `KC.N`, and should resolve to `KC.XXXXXXX`. This seems to be a parsing error, though I have not dug into the code further.

Thanks to Philippe Charest for reporting this bug in the Matrix chat.

pcharest2000, 2021-12-05:
I have dug into the issue and KC.NO is being parsed as an alpha,  changing the order of the parsing logic resolves the issue, this by testing first for the special case of KC.NO or XXXXXX:

'''Python
class KeyAttrDict(AttrDict):
    def __getattr__(self, key, depth=0):
        if depth > 1:
            raise InfiniteLoopDetected()

        try:
            return super(KeyAttrDict, self).__getattr__(key)
        except Exception:
            pass

        # Basic ASCII letters/numbers don't need anything fancy, so check those
        # in the laziest way
        print(key)
        if key in ('NO', 'XXXXXXX'):
            print("in NO")
            make_key(
                names=('NO', 'XXXXXXX'),
                on_press=handlers.passthrough,
                on_release=handlers.passthrough,
            )
        elif key in ALL_ALPHAS:
            make_key(code=4 + ALL_ALPHAS.index(key), names=(key,))
            print("in alphas")

kdb424, 2021-12-05:
If it's tested, feel free to open a PR, and we'll get it reviewed, and hopefully merged.

pcharest2000, 2021-12-05:
It is tested,  but I'm not really  familiar with pull requests....  I have tried to do so  for the handler documentation but was not able to do so from the web interface...  was not able to pr on the main branch... will look into it tonight.

pcharest2000, 2021-12-06:
> 

PR completed !!!!

closed 2021-12-06 by klardotsh