KMKfw/kmk_firmware #292: kmk fails with "USB busy" on mac before booting the OS

mhorowitz, 2021-12-07:
I have been using kmk on a mac with a custom keyboard successfully for the last few weeks.
Recently, I discovered that the keyboard is not recognized after a reboot.
After some debugging (which was tricky, as I can't just look at the serial output when the connected computer hasn't booted yet), I discovered that `send_report` is throwing an exception:
```
Traceback (most recent call last):
  File "code.py", line 28, in <module>
  File "kmk//kmk_keyboard.py", line 504, in go
  File "kmk//kmk_keyboard.py", line 115, in _send_hid
  File "kmk//hid.py", line 128, in send
  File "kmk//hid.py", line 233, in hid_send
OSError: USB busy
```
This problem reproduces with a simple noop code.py which doesn't do anything interesting, and it repros on two different boards (itsybitsy m4 express, and itsybitsy rp2040).  The boot.py is a bit complex in order to disable usb storage so I could mount it RW to append failures to a log file in flash, and use keyboard keys to re-enable usb storage so I can read it from my development machine.  code.py is a simple 1x1 keyboard.  Both files are attached here (but renamed because github didn't like the .py extension).
[boot.txt](https://github.com/KMKfw/kmk_firmware/files/7665718/boot.txt)
[code.txt](https://github.com/KMKfw/kmk_firmware/files/7665763/code.txt)

To repro:
1. I'm using github 8063898eb279cebfcf3e2442162a5ced0585d8df
1. copy the files to a board running circuitpython 7.1.0-beta.1
2. find a mac which requires a password to decrypt its boot drive
3. turn it off
4. plug the board into the mac
5. turn on the mac.  don't type your password.
6. Type one key by shorting A2 and A3.
6. wait a few seconds for the led to flash green. this means an exception was raised, caught, written the error to its flash storage, and the script has exited.
7. repro 2: unplug to board, and plug it back in again.  Type one key again, wait until the led flashes green again.
8. boot your mac, or any other machine.
8. connect the board to your dev box.
9. tap the switches on the board to enter safe mode (or short A2 and MOSI during powerup)
10. `cat .../CIRCUITPY/log.py` to see the error as above.

You'll notice code.py has a `time.sleep()` right before `keyboard.go()`, so this isn't a trivial timing issue, but perhaps it's something more complex.  Also, it's possible it's a circuitypython issue, but I'm not sure how to isolate that.

Any suggestions?

kdb424, 2021-12-07:
Thanks for finding a way to finally get this log. We have known it was an issue for ages, but had no clue how to get a log. Hopefully someone can take a look into this soon as it's a long standing bug, but we at least have something to start chasing now.

mhorowitz, 2021-12-19:
Another data point: I also get the USB busy error when I plug the keyboard into my KVM switch (https://www.startech.com/en-us/server-management/sv431dpua)

xs5871, 2022-01-17:
This happend to me occasionally; then every single boot, possibly correlated with a BIOS update.
Here's how I "fixed" it: If there's an `OSError` when sending a report, re-run `post_init`.
```python
class USBHID(AbstractHid):
# [...]
    def hid_send(self, evt):
        # int, can be looked up in HIDReportTypes
        reporting_device_const = evt[0]

        try:
            return self.devices[reporting_device_const].send_report(
                evt[1 : HID_REPORT_SIZES[reporting_device_const] + 1]
            )
        except OSError:
            return self.post_init()
```
It's a dumb solution and doesn't address _why_ there's an error, but my KMK keyboard consistently registers during POST now.

mhorowitz, 2022-01-18:
I tried the change suggested by @xs5871.  This stops the crash (unsurprisingly), but the keyboard doesn't actually work.  I can only guess that whatever problem I'm having and whatever problem they are having are not the same.

kdb424, 2022-03-19:
Several changes have been introduced to limit this. Please open a new issue if this is still a problem.

closed 2022-03-19 by kdb424