KMKfw/kmk_firmware #294: RuntimeError: pystack exhausted

JuliusS123, 2021-12-18:
Hello,
While using a raspberry Pi Pico with Circuitpython on the newest version (7.0.0), I run into this error:
```
code.py output:                                                                                                            
Starting                                                                                                                   
Traceback (most recent call last):                                                                                         
  File "code.py", line 20, in <module>                                                                                     
  File "kmk/keys.py", line 355, in __getattr__                                                                             
  File "kmk/keys.py", line 43, in first_truthy                                                                             
  File "kmk/keys.py", line 317, in <lambda>                                                                                
  File "kmk/keys.py", line 36, in left_pipe_until_some                                                                     
  File "kmk/keys.py", line 60, in maybe_make_shifted_key                                                                   
  File "kmk/keys.py", line 692, in make_shifted_key                                                                        
  File "kmk/keys.py", line 355, in __getattr__                                                                             
  File "kmk/keys.py", line 43, in first_truthy                                                                             
  File "kmk/keys.py", line 288, in <lambda>                                                                                
  File "kmk/keys.py", line 36, in left_pipe_until_some                                                                     
  File "kmk/keys.py", line 55, in maybe_make_key                                                                           
  File "kmk/keys.py", line 671, in make_key                                                                                
RuntimeError: pystack exhausted 
```
while using this code:
```
print("Starting")

import board

from kmk.kmk_keyboard import KMKKeyboard
from kmk.keys import KC
from kmk.matrix import DiodeOrientation

keyboard = KMKKeyboard()

keyboard.col_pins = (board.GP5, board.GP6, board.GP7, board.GP8, board.GP9, board.GP10, board.GP11, board.GP12, board.GP13, board.GP14, board.GP15, board.GP16)
keyboard.row_pins = (board.GP3, board.GP2, board.GP4, board.GP1, board.GP0)
keyboard.diode_orientation = DiodeOrientation.COL2ROW
keyboard.keymap = [
        [  #QWERTY
        KC.GRV,  KC.N1,   KC.N2,   KC.N3,   KC.N4,   KC.N5,   KC.N6,   KC.N7,   KC.N8,   KC.N9,   KC.N0,   KC.BSPC,
        KC.TAB,  KC.Q,    KC.W,    KC.E,    KC.R,    KC.T,    KC.Y,    KC.U,    KC.I,    KC.O,    KC.P,    KC.DEL,
        KC.ESC,  KC.A,    KC.S,    KC.D,    KC.F,    KC.G,    KC.H,    KC.J,    KC.K,    KC.L,    KC.SCLN, KC.QUOT,
        KC.LSFT, KC.Z,    KC.X,    KC.C,    KC.V,    KC.B,    KC.N,    KC.M,    KC.COMM, KC.DOT,  KC.SLSH, KC.ENT,
        KC.PIPE, KC.LCTL, KC.LALT, KC.LGUI, LOWER,   KC.SPC,  KC.SPC,  RAISE,   KC.LEFT, KC.DOWN, KC.UP,   KC.RGHT
    ]
]

if __name__ == '__main__':
    keyboard.go()
```
which keymap is copy-pasted from https://github.com/KMKfw/kmk_firmware/tree/master/boards/boardsource/5x12
The rest is either just copy-pasted from the tutorial, or simple GPIO lines.
I have tried downgrading circuitpython, but unfortunately, this just affects the "ticks_ms" module.
I've also tried upgrading to the beta release (7.1.0-beta.3), but this doesn't seem to affect the error.
Any help or suggestions would be welcome, and I sincerely apologize if I'm just missing something super obvious.

JuliusS123, 2021-12-19:
well this is embarassing

closed 2021-12-19 by JuliusS123

minordemon, 2022-01-10:
I had the same problem and seem too have fixed it by removing the stack limit from the `boot.py`.

```python
import supervisor

# supervisor.set_next_stack_limit(4096 + 4096)
```

I discovered that there was a lot more stack size available while I was trying to build a custom version of circuitpy. So I have also 'frozen' kmk in the fimware, that is perhaps a reason why I have more stack, but I don't think so.

minordemon, 2022-01-17:
Ok, the stack size was not the definitive solution.

The code failed with a stack overflow on shifted keys. When the unshifted key was not already registered, the secondary lookup of the unshifted key runs out of memory. I solved it by preloading the unshifted key before the definition of the keymap. Similar to the RAISE and LOWER keys.

```python
# Preload unshifted keys (Avoids stack overflow)
tmp = KC.EQUAL # for KC.PLUS
tmp = KC.BSLS  # for KC.PIPE
del tmp
```

This code is specific for my keymap. Your code probably fails on the `KC.PIPE`, which requires a preload of the `KC.BSLS`.