KMKfw/kmk_firmware #323: Split Transport Abstraction

xs5871, 2022-02-08:
The split module has a lot of "if BLE:" and "if UART:" patterns in it, which will get worse with the addition of a PIO UART and someday a one-wire interface.
We should split off the transport implementation from the communcation protocol and use a common interface to send and receive from the split module.
@Tonasz already mentioned that in #320.

Tonasz, 2022-02-08:
It would also make it easier in a future to change a model of transferred between sides and the way of handling them. I'm not sure if this is still the same issue or the next one, but currently `split.py` only transfers matrix changes. This makes rgb sync and encoder on slave side impossible to work!

xs5871, 2022-02-08:
In order to keep some atomizity I would categorize that as the next step.
It would definitely make it easier to have a transport agnostic report. We should keep that in mind, and maybe design the abstraction layer to transfer any bytearray to future proof.
I also like the idea of designating multiple report types with headers, instead of mirroring the entire keyboard state. It's really not necessary to update for example LEDs every couple of milliseconds.