KMKfw/kmk_firmware #339: Example for disableing CDC/storage HID in boot.py

xs5871, 2022-02-25:
We should add an example for disableing CDC/storage HID in `boot.py` to the documentation, including setup for a boot mode key/switch.

kdb424, 2022-02-25:
Would happily take a PR for some docs on this.

justinesmithies, 2022-02-27:
I've just done this using my Pico TKL diode matrix to check for a particular key press at boot to enable maintenance mode and it works a treat. Happy to share my working example.
[boot.py](https://paste.sr.ht/~justinesmithies/c2013762698dd1560f6f11b9a10bddc893854a47)

kdb424, 2022-02-27:
If you could write that up in some sort of a readme style thing with the "required changes", and open a pull request, we'll surely get that in.

HBBisenieks, 2022-03-15:
I believe this has been solved by #359 and #361

closed 2022-03-19 by kdb424