KMKfw/kmk_firmware #347: Rotary encoder divisor

mbellitti, 2022-03-01:
I am trying to add a rotary encoder to my build. The model I am using has 24 retents but 12 cycles per rotation, so with the default settings KMK picks up every other "click" (rather than all of them).

The problem is solved in `rotaryio` by setting the `divisor` flag to `2` (see [here](https://github.com/adafruit/circuitpython/pull/5468)). Is there something similar in the current implementation of KMK?

kdb424, 2022-03-01:
Try changing the movement from 1 to 2 on lines 50, 56, and 58 (make that -2), and let us know how that works.

mbellitti, 2022-03-05:
I've played around a little bit with the encoder module and I understand what the problem is. 

The pin output follows the sequence 00>01>11>10 when turned in one direction and 11>01>00>10 when turned in the other. The retents (the stable positons of the encoder, the "clicks") are on both 00 and 11, but the current version of the code only updates when the new state is 11 (see [here](https://github.com/KMKfw/kmk_firmware/blob/067b7d2e5fc3ea963fa85806974404019de443fe/kmk/modules/encoder.py#L61)). 

I've modified the code to make it work in my case, but I'm seeing some bouncing when I turn the wheel too fast, which is really annoying. 

My solution looks like this 
```python
    def update_state(self):
        # Rotation events
        new_state = (self.pin_a.get_value(), self.pin_b.get_value())

        if new_state != self._state:

            if new_state[0] == new_state[1]:
                print(new_state, self._state,end="\n\n")
                if new_state[0] == self._state[0]:
                    self._direction = 1
                else:
                    self._direction = -1

                self._pos += self._direction
                self.on_move_do(self.get_state())

            self._state = new_state
```
and it mostly works. I think a more robust solutions would be to use `rotaryio`, so I think I will write a wrapper around that. Maybe we could have two version of the encoder module, one that uses rotaryio and one that doesn't. If I understand correctly the main reason not to use rotaryio is that it requires using two consecutive pins, but I think that's perfectly fine.

kdb424, 2022-03-05:
If you port to roteryio, we'll rope someone in to help merge the modules in if you need help, but we prefer, unless you can present a good case, to have only one module for a feature to prevent confusion. That doesn't mean we aren't interested in your issue, but solving it the right way for everyone, so we'll work with you on it. The discord/matrix is great for active help

mbellitti, 2022-03-05:
Sounds good, I'll join the discord. Where do I find the link? Also, what was the rationale for *not* using rotaryio?

kdb424, 2022-03-05:
The links are in the main readme. I don't know why to not use rotaryio. I'd ask the members there as the people that worked on it are active in chat.

mbellitti, 2022-03-05:
Perfect, thank you.

closed 2022-03-05 by mbellitti