KMKfw/kmk_firmware #354: keycoords not matching physical layout

fluver, 2022-03-08:
While building a split keyboard i noticed that the keycoords followed a strange pattern.
In the matrix channel i was asked to open an issue to provide additional information.
I am using two Feather nRF52840 Express and 6*8 matrices, while the first row corresponds to the keycoords 0-6(left) and 7-13(right) the second row starts at 256, the third at 512 and so on.
I have been using the latest KMK version(4d1f9e487e97d18c23c8847d3a94796df704e9b7) and PR #328, my config files can be found here: https://github.com/fluver/keyb.
I tried to set split.split_offset to 0 or 48 but saw no difference, a print(len(keyboard.row_pins), len(keyboard.col_pins)) before keyboard.go() only shows the (6,8) i expected.
Any further advice on how to debug this?

xs5871, 2022-03-08:
I don't see how the behavior you're describing would even be possible with the latest version.
Do a fresh setup of KMK, without the PR at first, and test one side at a time. Left side should report key numbers 0-47, right side 48-95.

fluver, 2022-03-10:
I just tested it without the PR and it works with 0-48 and 48-95, although i could only test the left side since without #328 there is no communication between the two halves.
I guess this can be closed? 
(or becomes a bug report for the PR?)

fluver, 2022-03-12:
just wanted to comment that i applied the changes suggested in PR #328 to the current KMK master and everything works.
Key numbers correspond to 0-47+48-95, both halves communicate via BLE (tested shift+layer modifiers) and the left side(target) is connected via BLE to my PC.
Its beautiful.

closed 2022-03-14 by fluver