KMKfw/kmk_firmware #355: ESP8266 support?

jorgeluisrezende, 2022-03-08:
My question is, this library works on esp8266?

klardotsh, 2022-03-08:
No, [ESP8266 lacks USB HID and Bluetooth](https://forum.micropython.org/viewtopic.php?t=3640) and to my knowledge is not an appropriate microcontroller for keyboard building.

closed 2022-03-08 by klardotsh

jorgeluisrezende, 2022-03-08:
Exists an library to do this (https://github.com/cnlohr/espusb)(Espusb) do you think If I assemble it together I can use this keyboard firmware?

klardotsh, 2022-03-08:
You'd have to work with Adafruit upstream in https://github.com/adafruit/circuitpython to get `espusb` integrated as a HID provider, since [we just use Adafruit's `usb_hid` module](https://github.com/KMKfw/kmk_firmware/blob/master/kmk/hid.py#L1) for USB HID, which is implemented in the C source of CircuitPython itself.

kdb424, 2022-03-09:
> Generally speaking KMK is only tested on M4, nRF52840, and RP2040 chips, though I believe M0s have been used in split keyboards in the past, and there's a few examples of STM32 in the wild I think.

While this is true, I would never suggest it's even close to remotely supported, or recommended. It was a nightmare, and even as an owner of one that's just a dumb matrix scanner, I would not use that again.

jorgeluisrezende, 2022-03-09:
@kdb424 you mean that you will not use esp8266 again?

kdb424, 2022-03-09:
It was an M0, but it won't run KMK proper for ram usage alone. If you are willing to deal with all of the maintenance and firmware changes yourself, you are free to do so, but I'd personally recommend anything that klardotsh listed in terms of chips as I know that KMK will run without issue on any of those.