KMKfw/kmk_firmware #369: Stack overflow in maybe_make_shifted_key

rrotter, 2022-03-28:
Use of shifted keys produces a reproducible stack overflow in maybe_make_shifted_key.

Testing on d4e72b9 (current as of this writing), CircuitPython 7.2.3, on Pi Pico.

main.py:
```import board

from kmk.kmk_keyboard import KMKKeyboard
from kmk.keys import KC
from kmk.matrix import DiodeOrientation

keyboard = KMKKeyboard()

keyboard.col_pins = (
    board.GP3, board.GP4, board.GP5, board.GP6, board.GP7, board.GP8,
    board.GP9, board.GP10, board.GP11, board.GP12, board.GP13, board.GP14, board.GP15,
)

keyboard.row_pins = (
    board.GP16,
    board.GP17,
    board.GP18,
    board.GP19,
    board.GP20,
)
keyboard.diode_orientation = DiodeOrientation.ROW2COL

keyboard.keymap = [
    [KC.LABK,  KC.RABK,  KC.LPRN,  KC.RPRN,  KC.CIRC,  KC.DLR,   KC.GRAVE, KC.HASH,  KC.EXLM,  KC.SLASH, KC.LBRC,  KC.RBRC,  KC.BSLASH,
     KC.NO,    KC.Q,     KC.W,     KC.F,     KC.P,     KC.B,     KC.TAB,   KC.J,     KC.L,     KC.U,     KC.Y,     KC.MINUS, KC.EQUAL,
     KC.ESC,   KC.A,     KC.R,     KC.S,     KC.T,     KC.G,     KC.BSPC,  KC.M,     KC.N,     KC.E,     KC.I,     KC.O,     KC.QUOTE,
     KC.LSFT,  KC.Z,     KC.X,     KC.C,     KC.D,     KC.V,     KC.NO,    KC.K,     KC.H,     KC.COMMA, KC.DOT,   KC.UP,    KC.RSFT,
     KC.NO,    KC.NO,    KC.NO,    KC.LALT,  KC.LCMD,  KC.ENTER, KC.SPACE, KC.NO,    KC.RCMD,  KC.RALT,  KC.LEFT,  KC.DOWN,  KC.RIGHT,],
]

if __name__ == '__main__':
    keyboard.go()
```

100% reliably produces the following stack overflow (yes, irrespective of using the recommended settings in boot.py):
```
Traceback (most recent call last):
  File "main.py", line 50, in <module>
  File "kmk/keys.py", line 346, in __getattr__
  File "kmk/keys.py", line 39, in first_truthy
  File "kmk/keys.py", line 308, in <lambda>
  File "kmk/keys.py", line 32, in left_pipe_until_some
  File "kmk/keys.py", line 56, in maybe_make_shifted_key
  File "kmk/keys.py", line 683, in make_shifted_key
  File "kmk/keys.py", line 346, in __getattr__
  File "kmk/keys.py", line 39, in first_truthy
  File "kmk/keys.py", line 279, in <lambda>
  File "kmk/keys.py", line 32, in left_pipe_until_some
  File "kmk/keys.py", line 51, in maybe_make_key
  File "kmk/keys.py", line 662, in make_key
RuntimeError: pystack exhausted
```

Any shifted keycode that isn't a on a number key produces the same result. The solution I found was refactoring keys.py to reduce the stack depth when calling maybe_make_shifted_key. I'll send a PR to address this shortly.

xs5871, 2022-03-30:
Your PR works. Let's get that out of the way. It improves stability without being disruptive, and that's appreciated. I'm not against merging it. That's why I'm not commenting on the PR, but here.

To me, after looking into this, it looks more like a hot-fix than a solution.
What I was thinking is: In the long run, maybe we should solve this issue on a deeper level and rethink how shifted keys are generated.

kdb424, 2022-03-31:
If it's not hurting most users, I encourage solving this the right way if there is a more correct way. I'm happy to merge if this is wide spread and is needed for the majority of people. There's a lot at play with pystack, especially when we have many boards that need special stack sizes set at boot time. I'm likewise not against merging, but want to address the whole issue, and not only bandage this while we see an actual issue at hand potentially.

rrotter, 2022-03-31:
Thanks. I'm happy to rework the key generation code more thoroughly. I started with the minimal change to avoid rocking the boat. As it happens, I've managed to blow the stack again even with this fix in place, so it might be worthwhile to remove a lot of the nested function calls.

xs5871, 2022-03-31:
I think I found a very easy solution: [fix-shiftedkey-pystack-exhaust](https://github.com/xs5871/kmk_firmware/tree/fix-shiftedkey-pystack-exhaust)
I don't want to open up a competing PR or anything like that. Feel free to check out my attempt and use in any kind you see fit, or not.

rrotter, 2022-04-02:
That's a better solution anyway. I'd say just go ahead and merge what you've got.

PR #378 (2022-04-03): fix pystack exhaustion caused by make_shifted_key.
Resolves #369:
This removes multiple layers of indirection, effectively making
`make_shifted_key` only as heavy on the stack as the reguler `make_key`.
closed 2022-04-03 by kdb424