KMKfw/kmk_firmware #372: Split logic missing offsets for NativeKeypadScanner

levpopov, 2022-03-29:
NativeKeypadScanner is required for direct GPIO boards, but currently does not work for split non-matrix keyboards (like Ferris Sweep). 

Both left and right halves report same key codes since offset is not factored in when reporting events. To fix, `scan_for_changes` should be updated as follows:

```
class NativeKeypadScanner(Scanner):

   ...

    def scan_for_changes(self):
        '''
        Scan for key events and return a key report if an event exists.

        The key report is a byte array with contents [row, col, True if pressed else False]
        '''
        ev = self.curr_event
        has_event = self.keypad.events.get_into(ev)
        if has_event:
            return KeyEvent(ev.key_number + self.offset, ev.pressed)
```

xs5871, 2022-03-30:
You are correct. Would you like to make a PR?

xs5871, 2022-04-16:
resolved by #389

closed 2022-04-16 by xs5871