KMKfw/kmk_firmware #377: "RuntimeError: maximum recursion depth exceeded". When run sample code for board ESP-C3-32S

TuTanK5, 2022-04-02:
Hi all,

I'm trying to get KMK to work on board Ai Thinker ESP-C3-32S.
So I was able to flash CircuitPython on the board and copy `/kmk/`, `/adafruit_ble/`, and `boot.py` to it.

I'm now trying out the sample code.
This board doesn't support USB HID so I remove all code related to USB HID (in `hid.py` & `kmk_keyboard.py`), and set the it to run in BLE mode in `code.py`. The `code.py` now looks like this:
```
print("Starting")

import board
from kmk.kmk_keyboard import KMKKeyboard
from kmk.keys import KC
from kmk.matrix import DiodeOrientation
from kmk.hid import HIDModes

keyboard = KMKKeyboard()

keyboard.col_pins = (board.IO4)
keyboard.row_pins = (board.IO5)
keyboard.diode_orientation = DiodeOrientation.COL2ROW

keyboard.keymap = [
    [KC.A,]
]

if __name__ == '__main__':
    keyboard.go(hid_type=HIDModes.BLE, ble_name='KMKeyboard')
```

When trying to run this code I got this in REPL output:
```
Starting
RuntimeError: maximum recursion depth exceeded
```

After some digging, I tried setting higher stack limit in `boot.py` like below but the result is still the same (I tried 4096 * from 1 to 10) :
```supervisor.set_next_stack_limit(4096*10)```

My commit SHA:
```
from kmk.consts import KMK_RELEASE; print(KMK_RELEASE)
copied-from-git
```

What goes wrong? And how could I fix this issue?

Thank you,
Tan

xs5871, 2022-04-04:
The error suggests that your efforts to remove the USB HID introduced an infinite loop.

TuTanK5, 2022-04-05:
Hi @xs5871,

Thank you a lot for looking into this. I've never thought about that possibility.

Do you think it is fixable? Any general direction that I should check to find/fix the issue?
I'm thinking of making a stub for usb_hid.

Thank you,

xs5871, 2022-04-05:
Nope, not clue. If you can solve it, we're happy to review and merge any improvements.

TuTanK5, 2022-04-06:
Yep, I'll look into it and see what I can do.
Thank you.

kdb424, 2022-04-16:
If this ends up an issue with KMK itself, feel free to comment here and I can re-open, or if offering a PR to fix, tag it.

closed 2022-04-16 by kdb424

LazaroFilm, 2023-05-12:
Has a fox been found for this? ESP-32 would be great for bluetooth only boards as some boards already have Bluetooth and battery modules embedded.