KMKfw/kmk_firmware #395: Shift Register Scanner fails on Go.

scndthe2nd, 2022-04-14:
Using the examples provided for Shift Register Scanner, the scanner fails, requesting 1 positional argument on 'go'. This argument is not requested with the keypad matrix or key scanner.

[image]

Tested on 2022/04/14 using latest build.
`print("Starting")

import board

from kmk.kmk_keyboard import KMKKeyboard
from kmk.keys import KC
from kmk.scanners.keypad import ShiftRegisterKeys
import time
import neopixel

from kmk.scanners import DiodeOrientation
keyboard = KMKKeyboard()

class keyboard(KMKKeyboard):
    def __init__(self):
        # create and register the scanner
        self.matrix = ShiftRegisterKeys(
            # require arguments:
            clock=board.d5,
            data=board.d4,
            latch=board.d3,
            key_count=8,
            # optional arguments with defaults:
            value_to_latch=True, # 74HC165: True, CD4021: False
            value_when_pressed=False,
            interval=0.02,
            max_events=64
        )

keyboard.keymap = [
    ["SW0",  "SW1",  "SW2", "SW3",
    "SW4",  "SW5",  "SW6", "SW7"],
]

KMKKeyboard.debug_enabled = True

pixel = neopixel.NeoPixel(board.NEOPIXEL, 1)

pixel.brightness = 0.01

if __name__ == '__main__':
    keyboard.go()

while True:
    pixel.fill((0, 0, 0))
`

xs5871, 2022-04-15:
There are a couple of errors in your configuration.
There's no positional argument to `go` because you didn't instanciate the keyboard object, but calling the method from the class definition. That's not the only issue in the code, though.
The matrix channel would be a good place to get help fixing your config first.

kdb424, 2022-04-16:
Closing as I saw this was a keymap problem, and not a KMK problem, as well as being discussed in matrix/discord. Feel free to reopen if this is actually a KMK issue.

closed 2022-04-16 by kdb424