KMKfw/kmk_firmware #398: Possibly add support for the Kinesis Advantage 360

kyancey, 2022-04-16:
I'm new to KMK and keyboard firmware, but not new to microcontrollers or mechanical keyboards. So I could be incorrect about this, but I think it would be very easy to add KMK support to the [Kinesis Advantage 360](https://drop.com/buy/kinesis-advantage360-split-ortho-ergonomic-keyboard/) when it comes out in a month or two.

It's advertised as supporting ZMK, and there is a [starter configuration here](https://github.com/PolarityWorks/zmk-config-adv360).

From Kconfig.board, you can see that both sides use the NRF52840-QIAA-R:

```
config BOARD_ADV360_LEFT
	bool "adv360_left"
	depends on SOC_NRF52840_QIAA

config BOARD_ADV360_RIGHT
	bool "adv360_right"
	depends on SOC_NRF52840_QIAA
```

The [left](https://github.com/PolarityWorks/zmk-config-adv360/blob/main/config/boards/arm/adv360/adv360_left.dts)and [right](https://github.com/PolarityWorks/zmk-config-adv360/blob/main/config/boards/arm/adv360/adv360_right.dts) dts files contain the gpio pins for the rows, columns, and led lights. There's some more info in [adv360.dts](https://github.com/PolarityWorks/zmk-config-adv360/blob/main/config/boards/arm/adv360/adv360.dtsi) that may be useful.

I think this keyboard could pretty much be loaded with KMK at launch, but as I said, I'm new to this, and I may be missing something important. I look forward to any thoughts you may have.

kdb424, 2022-04-16:
We would need a project like https://github.com/boardsource/bs-python to add the hardware support, as we run on python on top of the RTOS like Circuitpython or a fork like that. It would be possible, but is partially an upstream problem. Once that gets ported, adding the keyboard firmware portion would be within KMK, and should be easy.

kyancey, 2022-04-16:
Okay. So basically, you're saying I need to get CircuitPython working on the boards before I get KMK working.

From what I've been reading [here](https://www.hackster.io/wallarug/circuitpython-creating-custom-boards-8e979e), making a custom board definition doesn't seem hard. The info needed all seem to be in the current config files or something I could figure out. The keyboard won't be available for several months. So I'll revisit this when I can actually get one in my hands.

Thanks.

kdb424, 2022-04-16:
Correct. KMK lives on top of any Circuitpython based operating system. All of the info should be there to add support to both though yeah.

kyancey, 2022-04-16:
I found out from the developers that it's using a Holyiot-18010, an integrated antenna module for nrf52840. The dtsi file for zmk is [here](https://github.com/zephyrproject-rtos/zephyr/blob/586e26e8fc055f5bbbfb8c94b6f59363fd128c31/dts/arm/nordic/nrf52840_qiaa.dtsi) and [here](https://github.com/zephyrproject-rtos/zephyr/blob/586e26e8fc055f5bbbfb8c94b6f59363fd128c31/dts/arm/nordic/nrf52840.dtsi). Posting this here for future reference.

klardotsh, 2022-04-17:
Wow, I hadn't seen a new Advantage was coming out. That board is *slick* and could probably finally get me to part with my 4-years-old-and-counting Keebio Iris (albeit after desoldering the switches and replacing them with... anything but those Gateron Browns).... whenever it (maybe) comes back in stock. *sigh*. I always miss the cool stuff these days...

kyancey, 2022-04-18:
There was a limited drop that was supposed to come out by now, but with the slowdown in China, they've pushed back delivery to mid-July. After that, I assume they'll start general sales. I suspect the rest of us won't get one until August-September. You can solder your own switches, but they are offering a paid service through Upgrade Keyboards. Probably expensive though.

### More information

PolarityWorks was kind enough to give me the info about the three led lights on both halves:

> It's a chain of 3 ws2812b LEDs on pin 0.20 of the microcontroller. Same pin on both sides.