KMKfw/kmk_firmware #419: Docs style guide needed

kdb424, 2022-04-24:
If anyone has experience in creating a style guide, we would love to have one with the size we are at. If you are interested, feel free to open a PR and/or talk to us in the matrix/discord chat about it.

rrotter, 2022-04-25:
Maybe add a markdown linter to the ci?
[markdownlint](https://github.com/DavidAnson/markdownlint)

Came here to write this comment as a separate bug, but since you got here first, idk if my suggestion is part of this bug, a duplicate, or a separate issue.

kdb424, 2022-04-25:
Maybe I had a duplicate. Whoops. Ping me there, and I'll switch to using that if this is the primary topic.

rrotter, 2022-04-25:
No, you didn't do anything wrong. What I was trying to say: _I was about to_ create a bug requesting markdown linting, but added a comment here instead.

rrotter, 2022-04-25:
Started on this in #420