KMKfw/kmk_firmware #424: Issue when typing random keys very fast

masiullah, 2022-04-26:
I am getting the below error in the serial when I type keys very fast. I am using KB2040

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
main.py output:
Traceback (most recent call last):
  File "main.py", line 33, in <module>
  File "kmk/kmk_keyboard.py", line 424, in go
  File "kmk/kmk_keyboard.py", line 477, in _main_loop
  File "kmk/kmk_keyboard.py", line 115, in _handle_matrix_report
  File "kmk/kmk_keyboard.py", line 165, in _on_matrix_changed
  File "kmk/kmk_keyboard.py", line 181, in pre_process_key
KeyError: 6

masiullah, 2022-04-26:
Here is my main.py config that throws this error

from kb import KMKKeyboard
from kmk.keys import KC
from kmk.modules.split import Split, SplitSide, SplitType
import board
  

keyboard = KMKKeyboard()
keyboard.debug_enabled = False

split = Split(use_pio=True, data_pin = board.D1)

keyboard.modules = [split ]

keyboard.keymap = [
    [  # QWERTY
        KC.Q,           KC.W,           KC.E,           KC.R,           KC.T,           KC.Y,           KC.U,           KC.I,           KC.O,           KC.P,
        KC.A,           KC.S,           KC.D,           KC.F,           KC.G,           KC.H,           KC.J,           KC.K,           KC.L,           KC.QUOT,
        KC.Z,           KC.X,           KC.C,           KC.V,           KC.B,           KC.N,           KC.M,           KC.COMM,        KC.DOT,        KC.ENT,
                                        KC.MUTE,       KC.TAB,        KC.SPC,        KC.LSFT,     KC.BSPC,     KC.MUTE,
                                        KC.VOLU, KC.VOLD,                             KC.VOLU, KC.VOLD,
    ],
]

if __name__ == '__main__':
    keyboard.go()

xs5871, 2022-04-26:
I can reproduce that if I _really_ start mashing keys (~50/second). Wouldn't call that "typing", but still. At that point the MCU is so busy that the loop frequency drops to just above 50/s (yes, I measured that repeatedly).
According to these numbers it's expected to drop key events, but we should still have a guard in place and not just crash...

masiullah, 2022-04-28:
I am have faced this more often but it was close to 10 keys per second(I had serial console and debug enabled)
and once it fails, it repeatedly triggers the last key until the keyboard is disconnect

masiullah, 2022-04-28:
Earlier I used data pin and data pin 2 and it was crashing very frequently but now I moved on to use only data pin and it is not crashing and requires to type a lot of keys like you mentioned in your comment. I will go back to two data pins and see if I can reproduce again

kdb424, 2022-04-29:
This bug is caused by the buffer filling up, and garbage data being read over the line. We have a bug in splits again thanks to the header being removed. The slower the loop goes, the less keys you have to mash to produce this bug without the fix that was recently removed, but originally patched by me.

kdb424, 2022-05-01:
https://github.com/KMKfw/kmk_firmware/blob/bd87b278b8fdc3024c1615f12c7dbd09e2ba05eb/kmk/modules/split.py#L371=
This is the line that should prevent that crash. Change that to something lower and let me know if that stops the crash. There may be failed logic now that's also getting in the way, and that is actually throwing away required bytes leading to said crash as well. If it's a logical issue, it will need looked into further. If it's just too high, it would be a simple fix.

wildesteven, 2022-05-02:
I ran into this same problem.  I found out that we were modifying a single instance of `KeyEvent` and adding it to the `matrix_update_queue` every time.  With a split board, you can type two keys per cycle instead of just one, which makes it possible to put multiple references to the same `KeyEvent` in the queue.  The error above happens when you have a queued event from the current side, for example `<Event: key_number 3 released>` and you add another event `<Event: key_number 6 released>`.  This changes the first event into the second one leaving key 3 pressed and releasing key 6 twice, causing `KeyError: 6`.

closed 2022-05-04 by xs5871