KMKfw/kmk_firmware #426: Support combining chording with hold-tap

a8ksh4, 2022-04-27:
Using KC.MT() and KC.LT() with a key for hold/layer and tap/character blocks it from working with chords.  

I'm working on porting the artsey.io layout to kmk but it's a requirement to be able to use keys for chording as well as holding for layer change (and using chording in the subsequent layer).  

Thanks for your consideration!

xs5871, 2022-04-29:
Could provide some example code for us?

a8ksh4, 2022-04-29:
```
print("Starting")
import board

from kmk.kmk_keyboard import KMKKeyboard
from kmk.keys import KC
from kmk.scanners import DiodeOrientation
from kmk.modules.layers import Layers
from kmk.modules.combos import Combos, Chord, Sequence
combos = Combos()
from kmk.modules.modtap import ModTap
modtap = ModTap()
# optional: set a custom tap timeout in ms
# modtap.tap_time = 300
from kmk.modules.oneshot import OneShot
oneshot = OneShot()
# optional: set a custom tap timeout in ms (default: 1000ms)
# oneshot.tap_time = 1500

keyboard = KMKKeyboard()
#keyboard.debug_enabled = True

keyboard.modules.append(Layers())
keyboard.modules.append(combos)
keyboard.modules.append(modtap)
keyboard.modules.append(oneshot)

keyboard.row_pins = (board.D1,)
keyboard.col_pins = (board.A0, board.A1, board.A2, board.A3, board.D10, board.MOSI, board.MISO, board.CLK)
keyboard.diode_orientation = DiodeOrientation.COL2ROW
#keyboard.diode_orientation = DiodeOrientation.ROW2COL

S_NUM = KC.LT(1, KC.S)
#S_NUM = KC.MT(KC.S, KC.MO(1))

keyboard.keymap = [
    # BASE Layer
    [S_NUM, KC.T,  KC.R,  KC.A,
    #[KC.S, KC.T,  KC.R,  KC.A,
     KC.O, KC.I,  KC.Y,  KC.E,],

    # NUM Layer
    [KC.TRNS, KC.N3, KC.N2, KC.N1,
     KC.NO,   KC.N6, KC.N5, KC.N4],

    # # SYMBOL Layer

    # # NAV Layer

    # # MOUSE Layer
]

OS_LSFT = KC.OS(KC.LSFT, tap_time=None)
OS_LCTL = KC.OS(KC.LCTL, tap_time=None)
combos.combos = [
    #### Alpha Chords ####
    # A
    Chord((KC.O, KC.E),       KC.B),
    Chord((KC.Y, KC.E),       KC.C),
    Chord((KC.A, KC.R, KC.T), KC.D),
    # E
    Chord((KC.R, KC.A),       KC.F),
    Chord((KC.T, KC.R),       KC.G),
    Chord((KC.I, KC.E),         KC.H),
    # I
    Chord((KC.S, KC.T),         KC.J),
    Chord((KC.O, KC.Y),         KC.K),
    Chord((KC.I, KC.Y, KC.E),   KC.L),
    Chord((KC.O, KC.I, KC.Y),   KC.M),
    Chord((KC.O, KC.I),         KC.N),
    # O
    Chord((KC.O, KC.I, KC.E),   KC.P),
    Chord((KC.S, KC.T, KC.A),   KC.Q),
    # R
    # S
    # T
    Chord((KC.I, KC.Y),         KC.U),
    Chord((KC.S, KC.R),   KC.V),
    Chord((KC.S, KC.A),         KC.W),
    Chord((KC.S, KC.T, KC.R),   KC.X),
    # Y
    Chord((KC.S, KC.T, KC.R, KC.A), KC.Z),

    #### Numeric Chords ####
    Chord((KC.N1, KC.N2), KC.N7),
    Chord((KC.N2, KC.N3), KC.N8),
    Chord((KC.N4, KC.N5), KC.N9),
    Chord((KC.N5, KC.N6), KC.N0),
    Chord((KC.N2, KC.N4), KC.BSPC),

    #### Other Chords ####
    Chord((KC.O, KC.I, KC.Y, KC.E), KC.SPACE),
    Chord((KC.O, KC.R, KC.A), KC.ESC), #escape
    Chord((KC.A, KC.E), KC.ENT), #enter
    Chord((KC.O, KC.T, KC.R, KC.A), KC.TAB), #tab
    Chord((KC.E, KC.R), KC.BSPC),
    Chord((KC.E, KC.R, KC.T, KC.S), OS_LSFT),
]

if __name__ == '__main__':
    keyboard.go()

```

xs5871, 2022-04-30:
The combo module only sees `S_NUM`, not the resolved `KC.S`.
That is "intentional", or down to how the `process_key` pipeline works.

a8ksh4, 2022-05-14:
Okay, that's exciting.  I'll try subbing in S_NUM into the combos.

a8ksh4, 2022-05-15:
Works great.  Thx!

closed 2022-05-15 by a8ksh4