KMKfw/kmk_firmware #427: Automatic Modifier Chaining for OneShot Keys

ldsands, 2022-04-27:
So as recommended by @xs5871 on the matrix channel I used `Combos` to allow for chaining modifier keys while using `OneShot` keys. I even added it to `docs/oneshot.md`. Unfortunately, there is a consistent application of the chained modifiers to the next two keys tapped/pressed after the `OneShot` modifier(s). This does stop it the `timeout` is reached, so it is still useful but still causes problems. I've tried both `Chord` and `Sequence` from `Combos` and I've tested the `per_key_timeout` using both `True` and `False`. I realize that just waiting for the `timeout` works or shortening the `timeout`. For me these options either makes the modifier chaining using this method unusable for some (less commonly used) shortcuts or I forget and the modifiers apply anyway.

Ideally I would like to be able to do something like the following when loading the `OneShot` module:

```python
oneshot = OneShot(chain_modifiers=True)
```

Implementing this would almost certainly involve the chaining code to be located in the `kmk/modules/oneshot.py` and/or `kmk/modules/holdtap.py` files. If this were to be done in `oneshot.py` file I think this could be done in the `process_key` function. Below I just have a `ModifierKey` if statement and a print function where I see this going but I'm not sure how to extract the modifier pressed (e.g. `KC.LCTL`) to then chain it and then return the chained modifiers to then apply it to the next key pressed/interruption or await the timeout.

```python
def process_key(self, keyboard, current_key, is_pressed, int_coord):
    '''Release os key after interrupting keyup.'''
    for key, state in self.key_states.items():
        if key == current_key:
            continue

        if state.activated == ActivationType.PRESSED and is_pressed:
            state.activated = ActivationType.HOLD_TIMEOUT
        elif state.activated == ActivationType.RELEASED and is_pressed:
            state.activated = ActivationType.INTERRUPTED
        elif state.activated == ActivationType.INTERRUPTED:
            if isinstance(current_key.meta.kc, ModifierKey):  # TESTCODE:
                print("current_key is a modifier key")  # TESTCODE:
            self.ht_released(key, keyboard)

    return current_key
```

The way I thought I would go about solving this would be to do a check for a modifier key when a `OneShot` key is interrupted. If the interrupted key is another `ModifierKey` then that modifier would be extracted added to the first `OneShot` modifier and then return the two (now) chained modifiers to then await another interruption or timeout. Does this seem like a good approach to the problem?

Other complications/considerations:

- If chaining more modifiers than just two the code will need to chain the new modifier with the previously pressed modifier to then apply to the next key pressed (and continue chaining until a non modifier key is pressed). I don't think this'll be as complicated once the rest is solved.
- I'm not sure how to determine which keycodes will need to be ignored when chaining modifiers. For example, I would want modifiers to ignore key presses that change layers. I think that ignoring layer changes is already implemented but perhaps some other class of keycode should also be ignored.

I think I can help with this but I cannot seem to figure out how to extract the modifier being pressed to then return it for chaining. Any help with that and I might be able to do the rest on my own assuming that this should be done in the `oneshot.py` file, that `holdtap.py` file confuses me.

Thanks in advance for all of your help and your work on KMK!

ldsands, 2022-05-19:
So I have an update on this. First off everything (except one item detailed below) seems to now be working with the modifier chaining. So whatever bug was affecting it seems to have been resolved. Something edited after May 6th fixed it. I don't think it was anything I did in my code since all I've done since then is to add OLED code to my `kb.py` and `main.py`.

ldsands, 2022-05-19:
I did find something of a workaround using chords, though it feels slower. Below is an example of me adding to the chords that I do/could use with a layer key.

```python
for layer_key in [MO_FUNCT, MO_NUM]:
    combos.combos.append(Chord((OS_LCTL, layer_key), OS_LCTL, timeout=750))
    combos.combos.append(Chord((OS_LSFT, layer_key), OS_LSFT, timeout=750))
    combos.combos.append(Chord((OS_LGUI, layer_key), OS_LGUI, timeout=750))
    combos.combos.append(Chord((OS_LALT, layer_key), OS_LALT, timeout=750))

    combos.combos.append(Chord((OS_LCTL, OS_LSFT, layer_key), OS_LCTL_LSFT, timeout=750))
    combos.combos.append(Chord((OS_LCTL, OS_LALT, layer_key), OS_LCTL_LALT, timeout=750))
    combos.combos.append(Chord((OS_LCTL, OS_LGUI, layer_key), OS_LCTL_LGUI, timeout=750))
    combos.combos.append(Chord((OS_LSFT, OS_LALT, layer_key), OS_LSFT_LALT, timeout=750))
    combos.combos.append(Chord((OS_LSFT, OS_LGUI, layer_key), OS_LSFT_LGUI, timeout=750))
    combos.combos.append(Chord((OS_LALT, OS_LGUI, layer_key), OS_LALT_LGUI, timeout=750))
```

Ideally, I still think it would be best to have an option built into the `OneShot` module that could allow for chaining modifier keys without having to resort to using the combos module.

xs5871, 2022-05-23:
The "feels slow" is another issue / missing combo feature / on my to-do list (an option to resolve a combo as soon as it matches, instead of waiting for the timeout).
Nevertheless, combos in this situation are verbose and tedious to setup. I agree that we should make this an option (or maybe default) within the `oneshot` module.

ldsands, 2022-05-23:
> The "feels slow" is another issue / missing combo feature / on my to-do list (an option to resolve a combo as soon as it matches, instead of waiting for the timeout).

Thank you! I think I may have said this elsewhere, so forgive me if this is repetitive, but `OneShots` (when I started using them with QMK) was probably one of the most effective things I did to reduce my wrist pain. Maybe I just use too many keyboard shortcuts :).

> Nevertheless, combos in this situation are verbose and tedious to setup. I agree that we should make this an option (or maybe default) within the `oneshot` module.

Yes, using combos is verbose, and I do think that setting this as the default, at least to me, makes sense.

I have tried to see if I could implement something like this myself but I have very little experience with oop. I'm mostly self-taught and I only do data science so I don't have to worry about classes very often. Most of KMK goes way over my head, so again thank you for working on this!

xs5871, 2022-06-07:
@ldsands:
from the QMK docs I couldn't figure out how the chaining "is supposed to" work (or even find a mention of it), and I don't feel like digging through the code right now.
If you could provide me with some insight:
* Would it be sufficient to chain only oneshot keys?
* similarly: is a check for special keys (mods, layer, etc.) necessary, for example: chain KC.LT, don't chain OS(KC.X)?
* Is this a global feature or per-key?

ldsands, 2022-09-21:
@xs5871 Sorry I didn't see this I must have accidentally dismissed it without realizing.

To answer your questions:

- I think it would be sufficient to chain only OneShot keys as long as they then allow the modifiers to apply to whatever key is pressed at the end of the sequence
- I would think that a check for special keys may be needed I can't really think of a situation I would use that would need to apply multiple oneshots that weren't either modifier keys (though applying them to a key pressed after a layer change would be useful)
- I think that setting this as a global setting could be useful but I personally would not want that. I usually only use OneShots on my base layer and then sometimes apply them to keys on another layer (for example tap OneShot control key and then the OneShot shift, change layer then hit home key to go to the beginning of a document). I'm not sure how you're thinking about using this globally but I have another layer that I specifically do not include any OneShot keys.
- Another thought I have as a possible improvement is that at least as I hold OneShots right now (with combos to allow for chaining modifiers) doesn't allow for the modifier to act on (for example) a mouse click until after a timeout is reached. I would prefer that if I hold the OneShot modifier that I can have the option of sending the modifier hold to the connected computer until I release the key or the combination of the OneShot (or chained OneShots) + the tapped key that finishes a sequence.

I hope those thoughts help.

Also, just to remind you so you're aware, multiple OneShot modifiers with Chords which is the workaround to this issue I've been using since you made that possibile is currently broken. I'm pretty sure [this issue is related](https://github.com/KMKfw/kmk_firmware/issues/585).

Thanks again for all of your work on this! Apart from it being broken at the moment, it has been working very well overall.

xs5871, 2022-10-22:
Hm. It's probably better (and easier) to add a chaining option to the oneshot module anyway.

ldsands, 2022-11-05:
> Hm. It's probably better (and easier) to add a chaining option to the oneshot module anyway.

I wouldn't know if it is easier to do it this way but I do think that would better at least from an ease of use perspective.

closed 2023-02-28 by xs5871

ldsands, 2023-03-15:
Just wanted to say thank you to @xs5871! This latest rewrite works great! It even allows me to use my mouse with the mod keys while I hold them down. In the past this wouldn't happen until after the timeout expired.

Again thank you, this really helps with my wrist pain!

PR #704 (2023-02-28): Feature oneshot stacking
resolves #427
resolves #535

Oneshot keys automatically stack and reset timeouts. I'm not going to implement an option to toggle that off, unless there's a real world application / request.