KMKfw/kmk_firmware #447: KC.MACRO_SLEEP_MS(ms), MACROS missing documentation

chemicalwill, 2022-05-21:
There is a keycode `KC.MACRO_SLEEP_MS(ms)` in [/docs/keycodes.md](/docs/keycodes.md) that points the user to "MACROS," which looks like it should point to [/docs/sequences.md](/docs/sequences.md). It appears that there is no documentation for this keycode in `sequences.md` or anywhere else in the repo.

kdb424, 2022-05-21:
I can confirm that this does exist, there are no docs for it, and only one board has used it. This should find a home being documented.

chemicalwill, 2022-05-21:
Sounds good. I will dig into it. 

[This is the only place I've found it so far.](https://github.com/KMKfw/kmk_firmware/blob/73e17d780a03e7eefbec4b18a8e370b14e599499/kmk/keys.py#L152) Could you point me in the direction of where it's implemented, and the board that used it?

chemicalwill, 2022-05-21:
Scratch that - I think I've found it. I'll get on it tonight or tomorrow.

klardotsh, 2022-05-24:
Yep, https://github.com/KMKfw/kmk_firmware/blob/5f1b9d2b14cb5b920020205d723a5ed374b0427f/boards/atreus62/main.py is the only board to use this currently, though I once upon a time used it in one of my feature-testing boards too.

There's tons of undocumented easter eggs and helpful tidbits in the Sequences bits of the codebase, some of which I've probably forgotten by now. It's been years.

closed 2022-06-06 by kdb424