KMKfw/kmk_firmware #450: Migrate to CircuitPython 7.3

ldsands, 2022-05-23:
This issue is more to put this on everyone's radar.

I haven't had a chance to try it out but CircuitPython 7.3 is out now and although I don't think there are any additions that'll be a big help to KMK they do have a couple of additions that could be useful.

Primarily of interest to me is the addition of f-strings. Also, they have a couple of additions that address "RP2040 PIO" I don't know if this will be useful at all or not.

xs5871, 2022-05-23:
KMK already works with 7.3.0 beta/rc-2 (I'm usually on bleeding edge).
Our sources have been using f-strings for a while (I don't know which ports didn't support it until now), and the background write for PIO isn't really necessary, since we're only writing a couple of bytes into the PIO buffer anyways. It could potentially save nano seconds for split users, if someone wants to tackle that.

ldsands, 2022-05-23:
Awesome! I didn't realize you were already using 7.3. I don't think that the PIO background is something I would take the time to try and figure out anyway. Should I close the issue then?

xs5871, 2022-05-23:
Let's leave it open for a couple of days. Maybe someone else has a comment or other input.

kdb424, 2022-05-23:
We have been using f-strings for ages AFIK. @klardotsh is actually the one that sent that upstream to Circuitpython/Micropython

xs5871, 2022-06-12:
Nothing else to discuss and 8.0 is on the horizon.

closed 2022-06-12 by xs5871