KMKfw/kmk_firmware #452: Bluetooth keycodes not present in keys.py

ambustion, 2022-05-25:
The keycodes for Bluetooth located in the [docs](https://github.com/KMKfw/kmk_firmware/blob/master/docs/keycodes.md#bluetooth-keys) don't actually do anything that I can see.  There is a function for BLE_REFRESH however which is not listed.

greyhatmiddleman, 2022-12-18:
Just a heads up it seems that some of these files mentioned in the following pull request were not merged with the most recent master: https://github.com/KMKfw/kmk_firmware/pull/167/files

andywarburton, 2025-10-01:
I have submitted a [pull request](https://github.com/KMKfw/kmk_firmware/pull/1125) to remove these from the codebase since its been... 3 years since this problem was identified.