KMKfw/kmk_firmware #463: Online/rendered version of docs on readthedocs.org

stefanfoulis, 2022-06-05:
Would it be desirable to have the docs nicely rendered and available somewhere online?
For example on https://readthedocs.org/

It is possible to set it up so that the docs are automatically built and updated on readthedocs whenever anything is added to the `master` branch.

Maybe related to #419 .

stefanfoulis, 2022-06-05:
I could help setting this up.

kdb424, 2022-06-05:
Kinda like this? http://kmkfw.io/docs/

stefanfoulis, 2022-06-05:
haha, wow. Those docs are very nice.
I totally missed them :man_facepalming: .

It hasn't showed up on any of my google searches. Even when searching for `kmk docs`, `kmk keyboard docs` or `kmk firmware docs` (github is most prominent in the results).

So maybe we could change this issue to read `Link to docs from README.md` :man_shrugging:  :grin: .

PR #477 (2022-06-16): Adds link to Documentation in README.md
and fixed spacing of shields.

fixes #463
commit 0f9eb196 (2022-06-16): Adds link to Documentation in README.md

and fixed spacing of shields.

fixes #463
closed 2022-06-16 by kdb424