KMKfw/kmk_firmware #466: Keyboard Reboots with Error

amerikiwi93, 2022-06-06:
Keyboard randomly fails with the following error `AttributeError: 'NoneType' object has no attribute 'on_press'` then reboots.

```
MatrixChange(ic=9 pressed=False)
KeyResolution(key=Key(code=1005, has_modifiers=None))
KMKKeyboard(debug_enabled=True diode_orientation=0 matrix=(<MatrixScanner object at 0x2000c6e0>,) unicode_mode=0 _hid_helper=USBHID(REPORT_BYTES=9) keys_pressed =set() coordkeys_pressed={10: None} hid_pending=False active_layers=[0] timeouts ={536854517: [<closure>], 536854400: [None], 536854402: [None], 536854379: [None]} )
KMKKeyboard(debug_enabled=True diode_orientation=0 matrix=(<MatrixScanner object at 0x2000c6e0>,) unicode_mode=0 _hid_helper=USBHID(REPORT_BYTES=9) keys_pressed =set() coordkeys_pressed={10: None} hid_pending=False active_layers=[0] timeouts ={536854517: [<closure>], 536854400: [None], 536854402: [None]} )
MatrixChange(ic=10 pressed=False)
KeyResolution(key=Key(code=1006, has_modifiers=None))
KMKKeyboard(debug_enabled=True diode_orientation=0 matrix=(<MatrixScanner object at 0x2000c6e0>,) unicode_mode=0 _hid_helper=USBHID(REPORT_BYTES=9) keys_pressed =set() coordkeys_pressed={} hid_pending=False active_layers=[0] timeouts={536854 517: [None]} )
MatrixChange(ic=7 pressed=True)
KeyResolution(key=Key(code=1003, has_modifiers=None))
KMKKeyboard(debug_enabled=True diode_orientation=0 matrix=(<MatrixScanner object at 0x2000c6e0>,) unicode_mode=0 _hid_helper=USBHID(REPORT_BYTES=9) keys_pressed =set() coordkeys_pressed={7: Key(code=1003, has_modifiers=None)} hid_pending=False active_layers=[0] timeouts={536854517: [None], 536854679: [<closure>]} )
MatrixChange(ic=8 pressed=True)
KeyResolution(key=Key(code=1004, has_modifiers=None))
KMKKeyboard(debug_enabled=True diode_orientation=0 matrix=(<MatrixScanner object at 0x2000c6e0>,) unicode_mode=0 _hid_helper=USBHID(REPORT_BYTES=9) keys_pressed=set() coordkeys_pressed={8: None, 7: Key(code=1003, has_modifiers=None)} hid_pe                                                                                                                                                             nding=False active_layers=[0] timeouts={536854517: [None], 536854679: [<closure>                                                                                                                                                             ]} )
MatrixChange(ic=9 pressed=True)
KeyResolution(key=Key(code=1005, has_modifiers=None))
KMKKeyboard(debug_enabled=True diode_orientation=0 matrix=(<MatrixScanner object at 0x2000c6e0>,) unicode_mode=0 _hid_helper=USBHID(REPORT_BYTES=9) keys_pressed =set() coordkeys_pressed={8: None, 9: None, 7: Key(code=1003, has_modifiers=None )} hid_pending=False active_layers=[0] timeouts={536854679: [<closure>]} )
MatrixChange(ic=7 pressed=False)
KeyResolution(key=Key(code=1003, has_modifiers=None))
KMKKeyboard(debug_enabled=True diode_orientation=0 matrix=(<MatrixScanner object at 0x2000c6e0>,) unicode_mode=0 _hid_helper=USBHID(REPORT_BYTES=9) keys_pressed =set() coordkeys_pressed={8: None, 9: None} hid_pending=False active_layers=[0] timeouts={536854692: [<closure>], 536854694: [<closure>], 536854679: [None]} )
MatrixChange(ic=10 pressed=True)
KeyResolution(key=Key(code=1006, has_modifiers=None))
KMKKeyboard(debug_enabled=True diode_orientation=0 matrix=(<MatrixScanner object at 0x2000c6e0>,) unicode_mode=0 _hid_helper=USBHID(REPORT_BYTES=9) keys_pressed =set() coordkeys_pressed={8: None, 9: None, 10: None} hid_pending=False active_layers=[0] timeouts={536854692: [<closure>], 536854694: [<closure>], 536854679: [None]} )
MatrixChange(ic=8 pressed=False)
KeyResolution(key=Key(code=1004, has_modifiers=None))
Failed to run process_key function in module:  'NoneType' object has no attribut e 'on_press' <ModTap object at 0x200096f0>
Traceback (most recent call last):
  File "code.py", line 119, in <module>
  File "kmk/kmk_keyboard.py", line 468, in go
  File "kmk/kmk_keyboard.py", line 521, in _main_loop
  File "kmk/kmk_keyboard.py", line 115, in _handle_matrix_report
  File "kmk/kmk_keyboard.py", line 165, in _on_matrix_changed
  File "kmk/kmk_keyboard.py", line 184, in pre_process_key
  File "kmk/kmk_keyboard.py", line 192, in process_key
  File "kmk/keys.py", line 458, in on_release
  File "kmk/modules/holdtap.py", line 124, in ht_released
  File "kmk/modules/holdtap.py", line 149, in send_key_buffer
AttributeError: 'NoneType' object has no attribute 'on_press'

Code done running.
```

crazyiop, 2022-06-06:
@amerikiwi93 You did something wrong in your keymap. Please post your full keymap and the definition of your custom key in it if any.

You seems to have not properly use the modtap module. [see here](https://github.com/KMKfw/kmk_firmware/blob/master/docs/modtap.md)

Note that you only une 'modtap /ModTap` to include it in the modules list. NOT directly in the keymap. The associated key are still KC.xx.

crazyiop, 2022-06-06:
I did not sucessfully reproduced that... And see nothing wrong in your code. Is it reproducible on your side ? Do you remember what key/combination triggered the error ?

amerikiwi93, 2022-06-07:
It happened again. It was some combination of keys 
- 46 : BSPC_NUM `BSPC_NUM = KC.LT(NUM,KC.BSPC,prefer_hold=True,tap_interrupted=False)`
- 34 :  N_RSFT  `KC.MT(KC.N, KC.RSFT,prefer_hold=False,tap_interrupted=True)`
- 9 : KC.N6

I have also had it happen with the same keys where the keyboard doesn't respect the layer tap and stays in the **NUM** layer. 

Is it possible that this is hardware related?

xs5871, 2022-06-07:
That's a bug.
Fixing the crash is trivial, but you'd still get unexpected behavior. We probably have to step up our unit tests to get this fixed properly.

amerikiwi93, 2022-06-07:
Let me know if there is anything you need me to do. I'd like to help if possible.

xs5871, 2022-06-10:
@amerikiwi93 yes, you can help! (even more than opening an issue and providing a descent description, very much appreciated btw)
You could check out the draft PR and tell me wether this fixes the issue. Unfortunately, in unit-tests I couldn't reproduce the exact thing you described.

amerikiwi93, 2022-06-11:
@xs5871 I have installed #475 and with all my testing the issue is fixed. nThanks!

PR #475 (2022-06-11): fix inconsistencies with chained holdtap keys
May resolve #466.
Refactors some of the `HoldTap` code that caused unaccounted-for side-effects when more than one ht is active at once + more unit-tests.
In draft until verified by the issues OP.
closed 2022-06-11 by kdb424