KMKfw/kmk_firmware #489: PEG RGB_TOG doesn't sync on splits

kdb424, 2022-06-24:
The RGB module for PEG doesn't toggle RGB on both sides of a split. It turns off the side that is plugged into USB, but it doesn't seem to send to the secondary half.

kdb424, 2022-06-24:
On further research, this may be a PEG issue as both sides needed forced matched. There should be a new keycode though to force sync them. If they get out of alignment, toggle will leave them desynced as both with toggle to opposite states. RGB_OFF and RGB_ON should be valid keycodes to enable a user to force sync them

klardotsh, 2022-06-24:
Alternatively, RGB state could be periodically (when?) synced without a "fake" keycode in the process: maybe it's time for the split protocol to include header bytes to turn into more of a proper RPC protocol? (that'd take @xs5871 input)

xs5871, 2022-06-28:
Proper RPC / two-way communication isn't going to happen (from my side) until the split code has been de-spaghettified.

kdb424, 2022-08-09:
I made that quite a mess. That needs a total overhaul honestly. I'll take responsibility for that. It's first on my list of things to do when I have time to actually sit down and crack out a ton of code.

kwmlodozeniec, 2022-09-11:
I'm experiencing the same issue without using peg, just straight up KMK and the default RGB extension. I'm using KB2040 controller on a CRKBD.

Boshimanu, 2022-11-13:
> @kwmlodozeniec 
Same problem here, maybe I'm missing something, but the right side, powers on (RGB turn on) before the left side. The connection is on the left side. On the documentation it is not explain ho to do a configuration. They only say that you need to configure per side, but nothing on how to have sync configuration.

AshenColors, 2023-03-23:
Also experiencing with a fresh Corne and KB2040 with the (non-PEG) RGB extension. Could potentially be a KB2040-specific issue, or one unrelated to the PEG code that should be split into its own issue? I'm still trying to wrap my head around the codebase so I'm not quite sure yet.

hxse, 2023-09-23:
Corne with helios-rp2040 encountered similar problems.
https://kmkfw.zulipchat.com/#narrow/stream/358347-general/topic/rgb.20only.20reactive.20on.20one.20side