KMKfw/kmk_firmware #513: Document a "bare minimum rescue" KMK setup for space-constrained boards

klardotsh, 2022-07-14:
Part two of the "Josh had to rebuild his keyboard" saga: the discovery that the KMK codebase is so big, it can't be copied in full to even a NiceNano (which, back in the day, was one of the biggest MCUs we supported - certainly a far cry from the Pyboards of yore). In fact, I had to do all sorts of wild stuff like remove comments from various source files and manually stitch together a minimum file structure to get the thing to fit. It occurs to me that such a "minimum viable product" document could be helpful to those trying to bring KMK up on other constrained systems. Here's what it took to get my user keymap (klardotsh/iris) booted - it's still not the truly minimum product, as I use things like media keys and splits and blah-blah-blah, but it's a starting point.

```
(woods) ~  » find /mnt -type f -not -name '.*'
/mnt/.fseventsd/no_log
/mnt/kb.py
/mnt/boot_out.txt
/mnt/main.py
/mnt/kmk/kmktime.py
/mnt/kmk/transports/pio_uart.py
/mnt/kmk/transports/__init__.py
/mnt/kmk/kmk_keyboard.py
/mnt/kmk/hid.py
/mnt/kmk/__init__.py
/mnt/kmk/keys.py
/mnt/kmk/types.py
/mnt/kmk/consts.py
/mnt/kmk/extensions/__init__.py
/mnt/kmk/extensions/media_keys.py
/mnt/kmk/utils.py
/mnt/kmk/scanners/__init__.py
/mnt/kmk/scanners/keypad.py
/mnt/kmk/modules/tapdance.py
/mnt/kmk/modules/holdtap.py
/mnt/kmk/modules/__init__.py
/mnt/kmk/modules/split.py
/mnt/kmk/modules/layers.py
/mnt/kmk/key_validators.py
/mnt/kmk/handlers/sequences.py
/mnt/kmk/handlers/__init__.py
/mnt/kmk/handlers/stock.py
```

kdb424, 2022-07-14:
It would help a ton if the upstream image actually shipped a not broken image that locks away most of the storage that the user can't access, leading to... Literally this.

xs5871, 2022-07-14:
Hint to the person implementing this documentation: we also provide a compile target for mpy_cross:
```bash
$ du -sh kmk
488K    kmk
$ make compile
$ du -sh .compiled/kmk
228K    .compiled/kmk
```

LukeDRussell, 2022-08-02:
Getting Nice!Nano working [is in the docs](https://github.com/KMKfw/kmk_firmware/blob/master/docs/Officially_Supported_Microcontrollers.md#nicenano) 😝 .

I've thought of getting a Github Action to compile KMK to mpy and post a .zip for users, and even taking it a step further of getting KMK into the CircuitPython Community Bundle, so it can be installed with circup.

kdb424, 2022-08-09:
CircleCI or other would be better suited for getting compiles going again. We used to have that, but it broke for reasons I don't remember, and it's better than leaving github to own this project. If it can't move platforms at any time, we aren't making open source software, and we want this to remain fully open.

klardotsh, 2022-08-18:
Generally agreed that we should take as few dependencies on GH as possible, however, we do already have precedent of using GHA: https://github.com/KMKfw/kmk_firmware/blob/master/.github/workflows/test.yml is our lone CI workflow right now.