KMKfw/kmk_firmware #516: GESC broken

kdb424, 2022-07-15:
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
main.py output:
Traceback (most recent call last):
  File "main.py", line 43, in <module>
  File "kmk/kmk_keyboard.py", line 462, in go
  File "kmk/kmk_keyboard.py", line 515, in _main_loop
  File "kmk/kmk_keyboard.py", line 105, in _handle_matrix_report
  File "kmk/kmk_keyboard.py", line 155, in _on_matrix_changed
  File "kmk/kmk_keyboard.py", line 178, in pre_process_key
  File "kmk/kmk_keyboard.py", line 184, in process_key
  File "kmk/keys.py", line 444, in on_press
  File "kmk/handlers/stock.py", line 50, in gesc_pressed
  File "kmk/keys.py", line 71, in __getattr__
  File "kmk/keys.py", line 370, in __getitem__
  File "kmk/keys.py", line 35, in first_truthy
  File "kmk/keys.py", line 212, in <lambda>
  File "kmk/keys.py", line 28, in left_pipe_until_some
  File "kmk/keys.py", line 42, in maybe_make_mod_key
  File "kmk/keys.py", line 674, in make_mod_key
  File "kmk/keys.py", line 663, in make_key
RuntimeError: pystack exhausted
```
This doesn't seem to be an issue with any other key. I believe there's some logic here that causes this to fail. I'm not sure where to start looking though. This has been tested on several boards, and has even hit @klardotsh on his personal board.

klardotsh, 2022-07-15:
yeah, even weirder, the board doesn't auto-recover. if i hit the escape key (GESC), the board perma-crashes and I need to unplug-replug to get input back.

as a vim user you can imagine my current pain.

xs5871, 2022-07-16:
Hm. Doesn't happen for me (probably a combination of hardware and config).
I planned on finishing up #430 anyway, that should give us more breathing room on the stack.

For the time being try this: Initialize all the keys in `GESC_TRIGGERS` manually, so they don't have to be created the first time you press GESC.

As for the auto-recover: Is that something we should inform CP about? Could be either intentional or not, a surface level search didn't bring anything up on that.

kdb424, 2022-07-16:
Maybe him and I are somehow running a version that happens to trigger the bug that no longer happens? I'll sync, and report back.

kdb424, 2022-07-16:
Have tested again, still getting the error. Not sure where to go from here.

PR #527 (2022-07-22): Extract key generators
finishing up #430.
may resolve #516 (saves a bit of space on the stack), if not, makes it easier to fix afterwards.
closed 2022-07-22 by kdb424